hidglobal/digitalpersona-devices #2: Getting Error in Angular 7

navinsaini, 2020-02-25:
ERROR in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js
Module not found: Error: Can't resolve 'WebSdk' in '...\node_modules\@digitalpersona\devices\dist\es5.bundles'

a-bronx, 2020-02-26:
Hello! The WebSDK is an external dependency (see [Browser Support](https://hidglobal.github.io/digitalpersona-devices/)). The DigitalPersona WebSDK is currently a part of DP WebComponents product installation, and it is not yet published as an npm package, so adding @digitalpersona/devices does not pull this dependency automatically. We consider publishing the WebSDK as an npm package in future.

You can get the WebSDK either from the PD WebComponents installation , or you can just copy the WebSDK module [from the sample](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk). Copy it in your local "vendor modules" directory (not into `node_modules`!), add this directory into the module resolution list, then import it as a global module (see an [example](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/src/index.ts)).

closed 2020-02-26 by a-bronx

ArmandoPerdomo, 2020-11-30:
Hello, can you please guys share the solution in the angular way?

I try the following 

1. Install `npm install @digitalpersona/devices`
2. Copy [these files](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk) into my src/app/core/modules folder
3. In my angular.json I add the index.js like this 
```
"scripts": [
              "src/app/core/modules/WebSdk/index.js"
            ],
```
4. in my app.component.ts I do this
```
import {FingerprintReader} from "@digitalpersona/devices";
...

private reader: FingerprintReader;
constructor() {
  this.reader = new FingerprintReader();
}
```

But in my console still getting errors

```
ERROR in node_modules/@digitalpersona/devices/dist/typings/devices/cards/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
                        ~~~~~~
node_modules/@digitalpersona/devices/dist/typings/devices/fingerprints/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
                        ~~~~~~
node_modules/@digitalpersona/devices/dist/typings/devices/iwa/device.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
```

am I missing something?

ArmandoPerdomo, 2020-11-30:
I add in the app.module.ts

```
import './core/modules/WebSdk';
```

But i'm getting new errors

```
WARNING in ./src/app/core/modules/WebSdk/index.js
Module not found: Error: Can't resolve 'vertx' in 'D:\Armando\Proyectos\Cupo Express\cupo-expess-totem\src\app\core\modules\WebSdk'

ERROR in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js
Module not found: Error: Can't resolve 'WebSdk' in 'D:\Armando\Proyectos\Cupo Express\cupo-expess-totem\node_modules\@digitalpersona\devices\dist\es5.bundles'
Error from chokidar (D:\): Error: EBUSY: resource busy or locked, lstat 'D:\pagefile.sys'
    
    ERROR in node_modules/@digitalpersona/devices/dist/typings/devices/cards/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/fingerprints/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/iwa/device.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~

```