hidglobal/digitalpersona-devices #3: npm ru build not able to run on macos

co2nut, 2020-02-28:
getting error "Not Found - GET https://registry.npmjs.org/api-extractor - Not found"

a-bronx, 2020-02-28:
Hello!

Did you [install build tools](https://hidglobal.github.io/digitalpersona-devices/maintain/build.html)? The @micosoft/api-extractor and @microsoft-api-documenter should be installed as global packages:

`npm install -g yarn npx @microsoft/api-extractor @microsoft/api-documenter`  (notice the -g flag)

co2nut, 2020-02-28:
Installed an run the build successfully.Found the generated es6/es5 JavaScript files. But still not sure how to use it 🙁

a-bronx, 2020-02-28:
You do not need to build any of `digitalpersona/*` libraries by yourself, unless you clone it and do some customizations. You can just [add them to your application with npm](https://hidglobal.github.io/digitalpersona-devices/tutorial.html), add `import` and start using them.

closed 2020-02-28 by a-bronx

co2nut, 2020-02-28:
@a-bronx thanks for the reply. By the way, i tried to import { FingerprintReader, SampleFormat } from '@digitalpersona/devices' and run my application. It showing error Module not found: Can't resolve 'WebSdk'. am I missing something?

a-bronx, 2020-02-28:
The WebSDK is an external dependency (see [Browser Support](https://hidglobal.github.io/digitalpersona-devices/)). The DigitalPersona WebSDK is currently a part of DP WebComponents product installation, and it is not yet published as an npm package, so adding @digitalpersona/devices does not pull this dependency automatically. We consider publishing the WebSDK as an npm package in future.

You can get the WebSDK either from the DP WebComponents installation, or you can just copy the WebSDK module [from the sample](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk). Copy it in your local "vendor modules" directory (not into `node_modules`!), add this directory into the module resolution list, then import it as a global module (see an [example](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/src/index.ts)).

co2nut, 2020-02-28:
@a-bronx I tried, but when I start my application. I'm seeing error like :

[image]

fyi, I'm working on react project

a-bronx, 2020-02-28:
Do you import the WebSDK module into the server side code by chance? 

Both the `WebSDK` and `@digitalpersona/devices` are **browser-only** modules, not Node modules. The `WebSDK` module provides a secure communication channel between the browser and fingerprint/card readers, using a DigitalPersona Client Agent native (Win32) app running on the user's machine as an intermediate. 

So you must make sure that you import them in the client-side code only, like [this](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/src/index.html).

co2nut, 2020-02-29:
@a-bronx I tried to initialise a new react project and just add the WebSdk folder directly. When I try to run the project it shows me below error. 
[image]

Is there any dependencies that I'm missing? I created the github repo for this https://github.com/co2nut/react-digitalperosna. Please correct me if I'm wrong.

co2nut, 2020-02-29:
@a-bronx I just notice you were saying "DigitalPersona Client Agent native (Win32) app running on the user's machine as an intermediate". Does it mean it only works on windows? Btw, I had updated the https://github.com/co2nut/react-digitalperosna to import from index.html but still getting the same error.

co2nut, 2020-03-02:
Tried to run in windows with driver installed. still getting the same error

limsocheat, 2020-05-14:
> Tried to run in windows with driver installed. still getting the same error

you found the solution? 
I got an error with WebSdk too. 

> This dependency was not found:   
> * WebSdk in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js

co2nut, 2020-05-14:
I use another way to get the reader work with edgejs. you may follow this [example](https://medium.com/@l2smasahiro/nodejs-with-digital-persona-finger-print-scanner-reader-integration-windows-fcb0a6dafa68)

limsocheat, 2020-05-14:
> I use another way to get the reader work with edgejs. you may follow this [example](https://medium.com/@l2smasahiro/nodejs-with-digital-persona-finger-print-scanner-reader-integration-windows-fcb0a6dafa68)

Thank You, do we need to install any SDK? like DigitalPersona SDK for windows? 
Sorry to ask, I am new to this project.

co2nut, 2020-05-14:
Yea, when i bought the device. It comes with the installation files. You may look from the official website for the installation files.

a-bronx, 2020-05-15:
> I use another way to get the reader work with edgejs. you may follow this [example](https://medium.com/@l2smasahiro/nodejs-with-digital-persona-finger-print-scanner-reader-integration-windows-fcb0a6dafa68)

If what you tried to accomplish was subscribing to fingerprint reader events in context of Node.js process, then yes, our library will not help you, because it is intended to be used in a web browser only (Chrome, Firefox, Edge, IE11; Electron should be fine too, but not tested), paired with either DigitalPersona Workstation or Lite Client running on a user's Windows machine.

omerts, 2021-12-14:
The WebSDK module doesn't compile in Create React App boilerplate due to ESLint errors mentioned above. AFAIK, it means the module's code has some real issues in it, such as undefined variables. 

> @a-bronx I tried to initialise a new react project and just add the WebSdk folder directly. When I try to run the project it shows me below error. [image]
> 
> Is there any dependencies that I'm missing? I created the github repo for this https://github.com/co2nut/react-digitalperosna. Please correct me if I'm wrong.
> 
> By the way, I'm trying to setup a project using MacOs

a-bronx, 2021-12-15:
Try one of [these](https://stackoverflow.com/questions/44292520/unexpected-use-of-self-no-restricted-globals-react) solutions.