hidglobal/digitalpersona-devices #4: WebSdk dependency not found

limsocheat, 2020-05-14:
I tried to start project with NuxtJs. 
when I import @digitalpersona/devices, 
`import { FingerprintReader, SampleFormat } from "@digitalpersona/devices";`
it shows error: 
```
This dependency was not found:
* WebSdk in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js 
```

b22sk2, 2020-05-14:
I have a same issue  angular 8

a-bronx, 2020-05-15:
The WebSDK is an external dependency (see [Browser Support](https://hidglobal.github.io/digitalpersona-devices/)). The DigitalPersona WebSDK is currently a part of DP WebComponents product installation, and it is not yet published as an npm package, so adding @digitalpersona/devices does not pull this dependency automatically. We consider publishing the WebSDK as an npm package in future.

You can get the WebSDK either from the DP WebComponents installation, or you can just copy the WebSDK module [from the sample](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk). Copy it in your local "vendor modules" directory (not into `node_modules`!), add this directory into the module resolution list, then import it as a global module (see an [example](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/src/index.ts)).

limsocheat, 2020-05-15:
> The WebSDK is an external dependency (see [Browser Support](https://hidglobal.github.io/digitalpersona-devices/)). The DigitalPersona WebSDK is currently a part of DP WebComponents product [...]

I already imported the WebSdk, but I still get more errors:
1. navigator is not defined
[image]
2. Can't resolve 'vertx' in ./assets/WebSdk/index.js  
[image]
3. This dependency was not found: 
[image]

FYI: I am developing project with NuxtJS, VueJS framework. 
[image]

Will this package work with VueJS? 
Do I need to install any other package/SDK to get it work? 

Please help.

a-bronx, 2020-05-15:
Given the "`navigator` is not defined" error I suspect that the `import` statement works not as you expected, and somehow it lands the package on a server side, where `navigator` is indeed not defined. I'm not familiar with NuxtJS internals and didn't try to use it in VueJS, but essentially the effect of you import must be same as if your HTML page contained 

```javascript
<script src="./scripts/WebSdk/index.min.js"/>`
<script src="./scripts/@digitalpersona/devices/dist/es5.bundles/index.umd.min.js" />
```

Can it be a server-side rendering that causes that error? I believe both `WebSdk` and `@digitalpersona/devices` must be excluded from SSR.

limsocheat, 2020-05-20:
> Given the "`navigator` is not defined" error I suspect that the `import` statement works not as you expected, and somehow it lands the package on a server side, where `navigator` is indeed not defined. I'm not familiar with NuxtJS internals and didn't try to use it in VueJS, but essentially the effect of you import must be same as if your HTML page contained
> 
> ```js
> <script src="./scripts/WebSdk/index.min.js"/>`
> <script src="./scripts/@digitalpersona/devices/dist/es5.bundles/index.umd.min.js" />
> ```
> 
> Can it be a server-side rendering that causes that error? I believe both `WebSdk` and `@digitalpersona/devices` must be excluded from SSR and imported as page globals instead. [Check this out](https://nuxtjs.org/faq/window-document-undefined), for example.

Thanks for your support.
Now I have switched to Angular 9. I have cloned your sample: https://github.com/hidglobal/digitalpersona-sample-angularjs. But I still don't know the workflow. 

My goal is to enroll user's fingerprint with Digital Persona U.A.U 4500, and save it to mysql database. And I want to to authenticate user's fingerprint which stored in database. 

1. Is it possible with this package and other (core, service, enrollment, authentication)? do you have step by step guideline. 
2. I found this document on your website: https://devportal.digitalpersona.com/U.are.U%20SDK%20Developer%20Guide.pdf. I wonder how I can get javascript sdk while I cannot login to your Dev Portal. 
3. https://devportal.digitalpersona.com/ I want to register for Dev Portal, but when I send email to sales@crossmatch.com, it got bounce back email. Do you know how can I register and login to your Dev Portal?

best.

a-bronx, 2020-05-20:
A tutorial for this library can be found [here](https://hidglobal.github.io/digitalpersona-devices/tutorial.html). Documentation for other DPAM libraries can be found [here](https://hidglobal.github.io/digitalpersona-access-management-api/). Also note that the UareU SDK PDF you referenced describes an older Javascript SDK; this package is a replacement for the older SDK.

In simple terms, the `@digitalpersona/devices` library allows you to collect fingerprint samples in a browser. To use obtained fingerprint samples for enrollment/authentication, you need to send these samples to a fingerprint feature extraction and matching engine, which is separate from this library. There is no Javascript-based fingerprint extraction/matching API; all Javascript SDKs are used only to connect to fingerprint devices from a browser, collect fingerprint samples and encode them to JSON in order to send them to the server.

DigitalPersona UareU SDK provides the needed fingerprint engine (FingerJet). It has native API (C++) and managed API (Java, C#),  so you need to use any of them on your backend.
 
Alternatively, DigitalPersona Authentication Server provides fingerprint authentication using ActiveDirectory or LDS as a database; it also provides a number of web services (Web Enroll, Web Auth, Web Policies, Web Claims etc).  The  `@digitalpersona/authentication` and `@digitalpersona/enrollment` provide clients for these web services. If you do not use DP Authentication Server, then you do not need them.

So, you typical workflow would be:
* use the `@digitalpersona/devices` to capture fingerprint samples in a browser;
* if you use DigitalPersona AD/LDS Authentication Server, then use `@digitalpersona/enrollment` to send collected samples for enrollment; then use `@digitalpersona/authentication` to send collected samples for verification and obtaining a JSON Web Token (JWT). You cannot use MySql as a fingerprint storage though, as DP Auth Server works only with AD/LDS stores.
* if you want to use your custom credential storage, then you can use UareU SDK on your backend to process collected fingerprint samples.

limsocheat, 2020-05-21:
> A tutorial for this library can be found [here](https://hidglobal.github.io/digitalpersona-devices/tutorial.html). Documentation for other DPAM libraries can be found [...]

Thanks for prompt response. 
How about registration at https://devportal.digitalpersona.com. could you tell me how I can register?

a-bronx, 2020-05-21:
> How about registration at https://devportal.digitalpersona.com. could you tell me how I can register?

Send your request to michael.smith2@hidglobal.com.

closed 2020-05-21 by a-bronx

Isfan14, 2022-05-29:
> We consider publishing the WebSDK as an npm package in future.

Any update on this one?

a-bronx, 2022-05-29:
Not yet, sorry for a delay.

a-bronx, 2025-05-09:
A `@digitalpersona\websdk` package is now available, please try it.

You may also want to switch to `@digitalpersona\fingerprint` package (supports only fingerprints) and/or to `@digitalpersona\card` (supports only cards).

The `@digitalpersona\devices` may be declared obsolete in near future, and everyone will be encouraged to use those smaller and more targeted packages instead.