hidglobal/digitalpersona-devices #5: Error in Angular9

tomahawk10, 2020-06-12:
Hello,

I'm having issue implementing the library in Angular9. I created an instance of FingerprintReader     
 (like in this [example](https://hidglobal.github.io/digitalpersona-devices/tutorial.html)), run the application using ng serve and it prompts me this error.
[image]

I also tried to implement your instruction [here](https://github.com/hidglobal/digitalpersona-devices/issues/2#issuecomment-591203951) and it prompts another error regarding vertx
[image]

I hope you can help me in this matter. Thanks in advance.

a-bronx, 2020-06-13:
Hi,

Please check a [similar issue](https://github.com/hidglobal/digitalpersona-devices/issues/4), especially [this comment](https://github.com/hidglobal/digitalpersona-devices/issues/4#issuecomment-631441288).

At first glance it looks like you either run the WebSdk in a Node.js instead of a browser, or your target browser does not support promises. See the library's [requirements](https://hidglobal.github.io/digitalpersona-devices/index.html): it is a browser-only lib, and it requires browsers to support promises and websockets.

tomahawk10, 2020-06-14:
Hi @a-bronx,

Thanks for the quick response. I commented out the "import 'WebSdk';" in the dist/es6/devices/websdk/channel.js and added index.js and es6.bundles/index.umd.min.js  into the "script" section of my angular.json. 
[image]

The websdk and vertx error was gone but I do have communication error in the Console of Chrome
[image]

Is the device causing the issue? My device is Digital Persona U.are.U 4500

a-bronx, 2020-06-14:
Do you have the DigitalPersona Lite Client or DigitalPersona Workstation installed? They provide the "native" part of the communication channel between the browser and the fingerprint reader.

tomahawk10, 2020-06-15:
Where can I get DigitalPersona Lite Client or DigitalPersona Workstation?

a-bronx, 2020-06-15:
DigitalPersona Workstation is a part of [HID DigitalPersona](https://www.hidglobal.com/products/software/activid/digitalpersona) suite (which also includes DigitalPersona Authentication Server). If you're not DigitalPersona customer yet, then please contact HID sales team.

DigitalPersona Lite Client Windows installer can be downloaded for free using [this link](https://www.crossmatch.com/AltusFiles/AltusLite/digitalPersonaClient.Setup64.exe).

tomahawk10, 2020-06-23:
Hi @a-bronx - It works!!! Thanks!!

closed 2020-06-23 by tomahawk10