hidglobal/digitalpersona-devices #6: FingerPrint Enroll and Verification

ingrocha, 2020-08-04:
Hello I already get the data from the reader but I have been looking for how I can enroll and verify the sample but I have not found a way to assert it on a local server can you help me pls!

a-bronx, 2020-08-04:
Hello!

This library is a part of [DigitalPersona Authentication Management (DPAM) suite](https://hidglobal.github.io/digitalpersona-access-management-api/), so for enrollment and fingerprint matching you need the DigitalPersona Authentication Server with a configured [DPAM Web Services](https://hidglobal.github.io/digitalpersona-access-management-services/index.html).

If you're not DigitalPersona customer yet, then please contact [HID Global](https://www.hidglobal.com/crossmatch?utm_source=crossmatch.com&utm_campaign=crossmatch.com&utm_medium=301) sales team.

If you already have DigitalPersona Authentication Server, then you should use [digitalpersona/enrollment](https://github.com/hidglobal/digitalpersona-enrollment) module to enroll samples, and [digitalpersona/authentication](https://github.com/hidglobal/digitalpersona-authentication) to do identification/authentication. You can find usage examples [here](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/application/identity/tokens/fingerprints), and tutorials [here](https://hidglobal.github.io/digitalpersona-enrollment/tutorial.html) and [here](https://hidglobal.github.io/digitalpersona-authentication/tutorial.html).

Alternatively, you can try to deploy your own fingerprint matching engine. I have no details right now, but the fingerprint sample format should be compatible with ANSI/ISO fingerprint template standards.

closed 2021-01-17 by a-bronx

rangerisrael, 2023-10-13:
> Hello!
>
> This library is a part of [DigitalPersona Authentication Management (DPAM) suite](https://hidglobal.github.io/digitalpersona-access-management-api/), so for enrollment and fingerprint [...]

sir how I use DigitalPersona Authentication Server do i need to buy it?

a-bronx, 2023-10-15:
@rangerisrael Yes, DigitalPersona Authentication Server is a commercial product.