hidglobal/digitalpersona-devices #9: Hello, can you please guys share the implementation in the angular way?

ArmandoPerdomo, 2020-11-30:
Hello, can you please guys share the solution in the angular way?

I try the following 

1. Install `npm install @digitalpersona/devices`
2. Copy [these files](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk) into my src/app/core/modules folder
3. In my angular.json I add the index.js like this 
```
"scripts": [
              "src/app/core/modules/WebSdk/index.js"
            ],
```
4. in my app.component.ts I do this
```
import {FingerprintReader} from "@digitalpersona/devices";
...

private reader: FingerprintReader;
constructor() {
  this.reader = new FingerprintReader();
}
```

But in my console still getting errors

```
ERROR in node_modules/@digitalpersona/devices/dist/typings/devices/cards/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
                        ~~~~~~
node_modules/@digitalpersona/devices/dist/typings/devices/fingerprints/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
                        ~~~~~~
node_modules/@digitalpersona/devices/dist/typings/devices/iwa/device.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.

1 /// <reference types="WebSdk" />
```

am I missing something?

_Originally posted by @ArmandoPerdomo in https://github.com/hidglobal/digitalpersona-devices/issues/2#issuecomment-735875733_

ArmandoPerdomo, 2020-11-30:
I add in the app.module.ts

```
import './core/modules/WebSdk';
```

But i'm getting new errors

```
WARNING in ./src/app/core/modules/WebSdk/index.js
Module not found: Error: Can't resolve 'vertx' in 'D:\Armando\Proyectos\Cupo Express\cupo-expess-totem\src\app\core\modules\WebSdk'

ERROR in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js
Module not found: Error: Can't resolve 'WebSdk' in 'D:\Armando\Proyectos\Cupo Express\cupo-expess-totem\node_modules\@digitalpersona\devices\dist\es5.bundles'
Error from chokidar (D:\): Error: EBUSY: resource busy or locked, lstat 'D:\pagefile.sys'
    
    ERROR in node_modules/@digitalpersona/devices/dist/typings/devices/cards/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/fingerprints/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/iwa/device.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~

```

a-bronx, 2020-12-01:
Probably duplicate of #5.

ArmandoPerdomo, 2020-12-01:
I tried all the implementations specified there, but I'm still getting these errors

```
 ERROR in node_modules/@digitalpersona/devices/dist/typings/devices/cards/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/fingerprints/reader.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
                            ~~~~~~
    node_modules/@digitalpersona/devices/dist/typings/devices/iwa/device.d.ts:1:23 - error TS2688: Cannot find type definition file for 'websdk'.
    
    1 /// <reference types="WebSdk" />
```

ArmandoPerdomo, 2020-12-01:
I try to implement the SDK with the implementation described in the [Developer Guide](https://usermanual.wiki/Document/UareUSDKDeveloperGuide.292953544/html#pf21) in "the Javascript API"

But I think that is an old specification, maybe It would work to implement enrollment and authentication?

That's the only think that I have to do.

In the specification, you have a method that returns a sample from the user's finger, when you `startAcquisition()`

a-bronx, 2020-12-01:
Probably the Typescript compiler in your project cannot find the `modules\WebSdk\index.d.ts`. Try to add the path to `typings` in `tsconfig.json`.

The `import './core/modules/WebSdk';` should also work, just make sure there is no server-side rendering in the import point, because WebSdk is a browser-only library. You'll get errors is WebSdk somehow gets imported in the NodeJS context.

---
The documentation referred by the link is an older Fingerprint WebApi (`fingerprint.sdk.js`). It provides fingerprint acquisition only, but does not provide any methods for enrollment or authentication. You'll need to implement your own enrollment/authentication using the raw fingerprint samples, or use a DigitalPersona Web Services.

The `@digitalpersona/devices` library implements both the Fingerprint WebApi and Cards WebApi functionality, allowing to acquire fingerprints and card credentials. It is pretty similar to the older Fingerprint WebApi, so you can use any of them if all you need is to acquire fingerprint samples.

If you also need enrollment and authentication with DigitalPersona Web Services, you can use `@digitalpersona/enrolment` and `@digitalpersona/authentication`, and they have better integration with the `@digitalpersona/devices`.

ArmandoPerdomo, 2020-12-15:
Sorry for my late response, 

Great!, thanks for the support

the better solution was this (Angular 9) if anyone needs

1. Install npm install @digitalpersona/devices
2. Copy these files into my src/app/core/modules folder
3. Add `modues` folder into my typeRoots
```
"typeRoots": [
    "node_modules/@types",
    "src/app/core/modules"
 ],
```
4. Switch my ecma script target in my `tsconfig.json` to `"target": "es6",`
5. Add these scripts to my `angular.json` folder
```
"scripts": [
      "src/app/core/modules/WebSdk/index.js",
     "node_modules/@digitalpersona/devices/dist/es6.bundles/index.umd.min.js"
 ],
```
6. Then in my `app.component.ts` I add this and works
`
this.reader.enumerateDevices().then(console.log);
`
-----

Now I have another problem, I need to do the enrollment and the authentications with the sample users

Which requirements do I need to do this?

In your official documentation you have this method

```
private onSamplesAcquired = (event: SamplesAcquired) =>
    {
        try {
            const samples = event.samples;
            const api = new FingerprintsAuth( <service endpoint URL> );
            const token = await (
                this.identity ? 
                    api.authenticate(this.identity, samples):
                    api.identify(samples));
            );
            this.notifyOnToken(token);
        }
        catch (error) {
            this.handleError(error);
        }
    }
```

But what means this? `<service endpoint URL>` this has to be some integration with your services or something? 

My requirement its compare to fingerprints to authenticate the user, that's my general purpose 

Thanks

a-bronx, 2020-12-15:
>  `<service endpoint URL>` this has to be some integration with your services or something?

Answered [here](https://github.com/hidglobal/digitalpersona-authentication/issues/1#issuecomment-745543883)

> I need to do the enrollment and the authentications with the sample users

This task is out of scope of the `@digitalpersona/devices` library, as it's only purpose is to collect user credential data like fingerprint samples. To match fingerprints, you need a fingerprint matching engine. 

One solution is to use [DigitalPersona Authentication Server](https://www.hidglobal.com/products/software/activid/digitalpersona), which is a Windows service integrated with ActiveDirectory and/or LDS accounts. It includes [DigitalPersona Access Management Web Services](https://hidglobal.github.io/digitalpersona-access-management-api/) (DPAM) and corresponding Web APIs (like `@digitalpersona/authentication` and `@digitalpersona/enrollment`) to communicate with browsers. This is what provide `service endpoint` in the sample code you quoted.

Another option is to use any other fingerprint matching engine supporting one of [fingerprint template formats](https://github.com/hidglobal/digitalpersona-devices/blob/master/dist/api/docs/devices.sampleformat.md), and implement your own biometric authentication web service, backed with your own biometric data store and the matching engine of your choice, and with your own JS client library. In this case you won't be able to use `@digitalpersona/authentication` and `@digitalpersona/enrollment` APIs, as they are provided as a part of DPAM suite.

closed 2021-01-17 by a-bronx

ArmandoPerdomo, 2021-01-27:
Thank you for the response,

My solution was to do only the user sample capturing with the @digitalpersona/devices, and for the enrollment and authentication I made my solution in .NET with the SDK, receiving the input and outputting if the verification was successful

lbaclao, 2021-01-28:
Tried @ArmandoPerdomo's solution and I managed to make it work. But after trying

> 7. Then in my app.component.ts I add this and works
> `this.reader.enumerateDevices().then(console.log);`

I got this error from channel.js

`ERROR Error: Uncaught (in promise): Error: Communication failure.` 

Seems like I'm missing something

a-bronx, 2021-01-28:
Do you have a DigitalPersona Workstation or DigitalPersona Lite Client installed? See the https://github.com/hidglobal/digitalpersona-devices/issues/5#issuecomment-643812592

d-mance, 2021-03-05:
> Thank you for the response,
> 
> My solution was to do only the user sample capturing with the @digitalpersona/devices, and for the enrollment and authentication I made my solution in .NET with the SDK, receiving the input and outputting if the verification was successful

Your solution works fine but there is a problem if the build of the application is made with jenkins. Jenkins will download a fresh library with the import line not commented.

Is there any suggestion on how to manage that situation?

Abdullah0991, 2021-04-18:
Thank you very much @ArmandoPerdomo for the guide <3.

The lib is working fine for me with Angular 11, but I have 2 warnings:

> Warning: ./src/modules/WebSdk/index.js
> Module not found: Error: Can't resolve 'crypto' in 'C:\Users\user\Desktop\testpersona\src\modules\WebSdk'
> 
> Warning: ./src/modules/WebSdk/index.js
> Module not found: Error: Can't resolve 'vertx' in 'C:\Users\user\Desktop\testpersona\src\modules\WebSdk'

How to solve those warnings ?

a-bronx, 2021-04-19:
> The lib is working fine for me with Angular 11, but I have 2 warnings:

Where and when do you have these warnings? Make sure you load the WebSdk module only into the browser using a regular `<script>` tag, not via `import`. For your TS code to compile you need to import only typings,  so make sure  `.d..ts` file is in your tsc typings path. You may also need to declare an external global object `WebSdk` in your bundler.

Abdullah0991, 2021-04-29:
I am really sorry for the delay...

> Where and when do you have these warnings

Just after following @ArmandoPerdomo [guide](https://github.com/hidglobal/digitalpersona-devices/issues/9#issuecomment-745261282).

This is the service code:

```
import '../../../assets/modules/WebSdk';

import { Injectable } from '@angular/core';
import { FingerprintReader, SampleFormat } from '@digitalpersona/devices';
import { from, Observable } from 'rxjs';
import { finalize } from 'rxjs/operators';
import { DeviceInfo } from '@digitalpersona/devices/src/devices/fingerprints/device';

@Injectable({
  providedIn: 'root'
})
export class FingerprintPersonaService {
  private reader: FingerprintReader;

  constructor() {
  }

  public initReader(): Observable<void> {
    this.reader = new FingerprintReader(new WebSdk.WebChannelOptions({}));

    this.reader.onDeviceConnected = (event) => {
      console.log('onDeviceConnected', event);
    };
    this.reader.onDeviceDisconnected = (event) => {
      console.log('onDeviceDisconnected', event);
    };
    this.reader.onQualityReported = (event) => {
      console.log('onQualityReported', event);
    };
    this.reader.onSamplesAcquired = (event) => {
      console.log('onSamplesAcquired', event.samples[0]);
    };
    this.reader.onErrorOccurred = (event) => {
      console.log('onReaderError', event);
    };

    return from(this.reader.startAcquisition(SampleFormat.PngImage));
  }

  public destroyReader(): Observable<void> {
    return from(this.reader.stopAcquisition()).pipe(finalize(() => {
      console.log('Reader destroyed');
      delete this.reader;
    }));
  }

  public getReaderInfo(devUid: string): Observable<DeviceInfo | null> {
    return from(this.reader.getDeviceInfo(devUid));
  }
}
```

a-bronx, 2021-04-29:
@Abdullah0991 
1. do you have `index.d.ts` file in your `/assets/modules/WebSdk`?
2. As I mentioned, make sure the `WebSdk/index.js` is not bundled with your modules, but instead [loaded with `<script src="/assets/modules/WebSdk/index.js">` in your `index.html`](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/src/index.html). You may need to [declare the `WebSdk` as a "global/external" library in your webpack/rollup/etc config](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/master/build/webpack/prod.js#L39).
2. try to add the `import './assets/modules/WebSdk` in your top-level `index.ts` only. Your code does not really need any types from `WebSdk`, the import is needed only to introduce those global types for `@digitalpersona/devices`

Abdullah0991, 2021-05-01:
> do you have index.d.ts file in your /assets/modules/WebSdk?

Yes.

> As I mentioned, make sure the WebSdk/index.js is not bundled with your modules, but instead loaded with <script src="/assets/modules/WebSdk/index.js"> in your index.html. You may need to declare the WebSdk as a "global/external" library in your webpack/rollup/etc config.

I did and it works well without warnings. I did't need to declare the WebSdk as a "global/external" library (will be any expected errors later?) .

> try to add the import './assets/modules/WebSdk in your top-level index.ts only. Your code does not really need any types from WebSdk

I didn't try. I think it is not needed.

a-bronx, 2021-05-03:
> I did and it works well without warnings

Was the issue resolved in the end? To clarify: did you fix it just by loading WebSdk into a page via `<script>`? Or by fixing the bundling in some other way?

> (will be any expected errors later?).

It depends on the WebSdk itself, I cannot say for sure as I'm not a maintainer of WebSdk and never did such tests. If you have two copies of the lib, then in the best case a second copy just redefines the same globals (or bails if they are already defined), and only one of them will be used; the only downside will be an increased JS load size. In worst case different parts of your code may start using different instances of it, which may cause unexpected and hard to detect issues. To be on a safe side, I'd recommend to check bundles to ensure they do not contain WebSdk code.

Abdullah0991, 2021-05-03:
> Was the issue resolved in the end? To clarify: did you fix it just by loading WebSdk into a page via <script>?

Yes.

Abdullah0991, 2021-05-08:
> Thank you for the response,
> 
> My solution was to do only the user sample capturing with the @digitalpersona/devices, and for the enrollment and authentication I made my solution in .NET with the SDK, receiving the input and outputting if the verification was successful

Sorry for asking here, but how can I download the .NET SDK? I google it but can't find anything?

a-bronx, 2021-05-08:
https://dotnet.microsoft.com/download

Abdullah0991, 2021-05-09:
> https://dotnet.microsoft.com/download

Sorry for my ambiguous question, I was asking for PersonaDigital SDK for .Net environment!

a-bronx, 2021-05-10:
The DigitalPersona .NET SDK is a part of the DigitalPersona Native API, which is installed automatically with one of DigitalPersona client products:
* DigitalPersona AD Workstation or LDS Workstation
* DigitalPersona AD Kiosk or LDS Kiosk

The DigitalPersona SDK can also be purchased separately. Please inquire [HID customer service](https://support.hidglobal.com/customer-service) team.

mjturizom, 2022-06-24:
> Thank you for the response,
> 
> My solution was to do only the user sample capturing with the @digitalpersona/devices, and for the enrollment and authentication I made my solution in .NET with the SDK, receiving the input and outputting if the verification was successful

Hello Armando, have you an example of this implementation? 
I have tried to make the comparison but I have not found a way, what type of information do you send to the .net service?

lucindom, 2022-07-08:
we made a DLL to be able to use the .NET library, and from there we make the comparison. Then we call the DLL with electron-edge-js

ImWicho, 2023-03-17:
> Thank you for the response,
> 
> My solution was to do only the user sample capturing with the @digitalpersona/devices, and for the enrollment and authentication I made my solution in .NET with the SDK, receiving the input and outputting if the verification was successful

Do you have a documentation or something like that? I want to do the same, but I do not know anything about .NET

okothhillary, 2023-03-18:
Hello. What would you love me to do?

radhe65gupta, 2023-08-29:
> Do you have a documentation or something like that? I want to do the same, but I do not know anything about .NET

Hey can you achieve this?

armandoatpulpomatic, 2023-08-30:
Hello Guys, sorry for the late response
The solution was done in .NET with the SDK who provides digitalpersona, this SDK must be paid and it has its own documentation, the code snippet can't share here

lucindom, 2023-09-29:
https://github.com/hidglobal/digitalpersona-devices, https://github.com/ivanmiranda/digitalpersona, https://github.com/hidglobal/digitalpersona-sample-angularjs

sumanavya, 2024-12-09:
> Sorry for my late response,
> 
> Great!, thanks for the support
> 
> the better solution was this (Angular 9) if anyone needs
> 
> 1. Install npm install @digitalpersona/devices
> 2. Copy these files into my src/app/core/modules folder
> 3. Add `modues` folder into my typeRoots
> 
> ```
> "typeRoots": [
>     "node_modules/@types",
>     "src/app/core/modules"
>  ],
> ```
> 
> 4. Switch my ecma script target in my `tsconfig.json` to `"target": "es6",`
> 5. Add these scripts to my `angular.json` folder
> 
> ```
> "scripts": [
>       "src/app/core/modules/WebSdk/index.js"
>  ],
> ```
> 
> 6. Commented out the `import 'WebSdk' ` from the file `node_modules/@digitalpersona/devices/dist/es6/devices/websdk/channel.js` (this is some kinda bug or something?)
> 7. Then in my `app.component.ts` I add this and works
>    `this.reader.enumerateDevices().then(console.log);`
> 
> Now I have another problem, I need to do the enrollment and the authentications with the sample users
> 
> Which requirements do I need to do this?
> 
> In your official documentation you have this method
> 
> ```
> private onSamplesAcquired = (event: SamplesAcquired) =>
>     {
>         try {
>             const samples = event.samples;
>             const api = new FingerprintsAuth( <service endpoint URL> );
>             const token = await (
>                 this.identity ? 
>                     api.authenticate(this.identity, samples):
>                     api.identify(samples));
>             );
>             this.notifyOnToken(token);
>         }
>         catch (error) {
>             this.handleError(error);
>         }
>     }
> ```
> 
> But what means this? `<service endpoint URL>` this has to be some integration with your services or something?
> 
> My requirement its compare to fingerprints to authenticate the user, that's my general purpose
> 
> Thanks

I have  followed the same steps in my angular8 project.

1,I have installed the digitalpersona client.
2.Followed the same steps you have mentioned .
3. this.reader.enumerateDevices().then(console.log); does not returning anything 

In console log i am getting zoneawarepromise ,device i am not getting 
[image]

Please help me here ,may i know why im not able to get device list

a-bronx, 2025-05-09:
A `@digitalpersona\websdk` package is now available, please try it.

You may also consider to switch to `@digitalpersona\fingerprint` package (supports only fingerprints) and/or to `@digitalpersona\card` (supports only cards).

The `@digitalpersona\devices` may be declared obsolete in near future, and everyone will be encouraged to use those smaller and more targeted packages instead.