hidglobal/digitalpersona-devices #11: Convert Samples Acquired to Image format

deunrandy, 2020-12-19:
Hello.

I already implemented and captured samples from my U.are.U 4500 Fingerprint Reader on Angular 9 framework. Is there a way to convert the sample to image format?

Thanks!

a-bronx, 2020-12-19:
To get raw fingerprint images (BMP, JPEG2000/WSQ or PNG), provide a corresponding [`SampleFormat`](https://github.com/hidglobal/digitalpersona-devices/blob/5f174cd5ab4949a97de5f3e35af507fec03e1f04/src/devices/fingerprints/sample.ts#L4) value as a parameter of the [`FingerprintReade.startAcquisition()`](https://github.com/hidglobal/digitalpersona-devices/blob/5f174cd5ab4949a97de5f3e35af507fec03e1f04/src/devices/fingerprints/reader.ts#L175). Do not use `SampleFormat.Intermediate` for this purpose, as it is an internal format.

deunrandy, 2020-12-19:
Thanks @a-bronx. I used `SampleFormat.PngImage` and got array of string in return. Is this string in base64 format? I tried converting it into image but received a broken image instead.

a-bronx, 2020-12-19:
The `onSamplesAcquired` event handler receives a [`SamplesAcquired`](https://github.com/hidglobal/digitalpersona-devices/blob/5f174cd5ab4949a97de5f3e35af507fec03e1f04/src/devices/fingerprints/events.ts#L6) object, which contains an array of [`BioSample`'s](https://github.com/hidglobal/digitalpersona-core/blob/20dfa8d959479f5b2992f31b1ff8733d1089ff3e/src/biometrics/factor.ts#L110). The `BioSample.Data` is a Base64Url-encoded image data.

deunrandy, 2020-12-19:
Thanks for the help @a-bronx. Closing this issue

Abdullah0991, 2021-04-29:
Sorry to reopen this, but I can't get the image. (using `Angular 11`)

I use the PNG format: `reader.startAcquisition(SampleFormat.PngImage)`

In `onSamplesAcquired` callback, I get an array which has one element contains what looks like a base64 encoding, but when I tried to use it, it gives me a broken image. The base64 size is quite big ~220kb.

I use it in `img` element like this: `src=data:image/png;base64,` + `the base64`, Also tried an online base64 visualizer and it gives me the same result, broken image.

a-bronx, 2021-04-29:
@Abdullah0991, the [`SampesAcquired`](https://github.com/hidglobal/digitalpersona-devices/blob/master/src/devices/fingerprints/events.ts#L6) event returns an array of [`BioSample`](https://github.com/hidglobal/digitalpersona-core/blob/master/src/biometrics/factor.ts#L110) objects in its `samples` array property. The image should be in the `BioSample.Data` field, which is base64url-encoded.

May be the `BioSample`s in the `samples` array are base64url encoded as well, I cannot recall it from my head and cannot verify now. In this case you may need to decode twice: first base64url-decode to obtain the `BioSample` object as JSON string, then `JSON.parse` to deserialize it to a `BioSample` object, then base64url-decode the `BioSample.Data` to get the raw image as a Utf8 string and, may be, use `TextEncoder` to convert the string to a `Uint8Array`, if needed. 

To decode you can use the [`Utf8`](https://github.com/hidglobal/digitalpersona-core/blob/master/dist/api/docs/core.utf8.md) utility class from the `@digitalpersona/core` library (it should be readily available to your project as a dependency of the `@digitalpersona/devices`), specifically you'll need `Utf8.fromBase64Url` method. Try this code (it is not verified, writing from my head now):

```ts
import { Utf8 } from '@digitalpersona/core'
...
this.reader.onSamplesAcquired = (data) => {
    const rawImages: Uint8Array[]  = data.samples.map(sample =>
        TextEncoder.encode(
            Utf8.fromBase64Url(
                JSON.parse(
                   Utf8.fromBase64Url(sample)
                ).Data
            )
        )
    )
};
```

If `samples` are indeed base64url-encoded, then I may need to fix typings of the `SamplesAcquired` event from `samples: BioSample[]` to `samples: Base64UrlString[]`. Let me know if you can confirm this.

Abdullah0991, 2021-05-01:
I have tested the code, the first error is at `JSON.parse` since the return from `Utf8.fromBase64Url(sample)` isn't a valid json string.
[ See the result of  Utf8.fromBase64Url(sample)](https://ibb.co/wydb15c)

Abdullah0991, 2021-05-02:
Hi @a-bronx , I managed to get the job done but not as you expected.

If you see the result of `Utf8.fromBase64Url(sample)` you will notice that it looks like a binary PNG image data. So I tried convert it to a base64 using `window.btoa()` and it worked !! The result base64 string give me a valid image.

a-bronx, 2021-05-03:
So, just to elaborate, was it the `btoa(Utf8.fromBase64Url(sample))` that worked for you? If it indeed was, then you probably may just use `Base64.fromBase64Url(sample)`, because `Utf8.fromBase64Url()` is just a composition of `Base64.fromBase64Url()` and `atob()`. You'll save on a `btoa(atob())` roundtrip.

closed 2021-05-03 by a-bronx

Abdullah0991, 2021-05-03:
> So, just to elaborate, was it the btoa(Utf8.fromBase64Url(sample)) that worked for you?

Yes.

`const sampleImageUrl = 'data:image/png;base64,' + btoa(Utf8.fromBase64Url(sample));`

pdhung197, 2021-12-08:
> > So, just to elaborate, was it the btoa(Utf8.fromBase64Url(sample)) that worked for you?
> 
> Yes.
> 
> `const sampleImageUrl = 'data:image/png;base64,' + btoa(Utf8.fromBase64Url(sample));`

I use the solution for my code, but seem like no luck, or have something wrong in my code.
This is my code:
In SampleAcquired event: 
```
const { Data: rawData } = samples[0];
const sampleImageUrl = 'data:image/png;base64,' + btoa(Utf8.fromBase64Url(rawData));
this.imgString = sampleImageUrl;
```
In template: ```<img [src]="imgString" alt="" />```
And the result like this:
[image]

Is there any mistake from me that make the code not work? :(

Abdullah0991, 2021-12-08:
Try:
```
import { Base64 } from '@digitalpersona/core';
const sampleImageUrl = 'data:image/png;base64,' + Base64.fromBase64Url(sample);
```

pdhung197, 2021-12-08:
> Try:
> 
> ```
> import { Base64 } from '@digitalpersona/core';
> const sampleImageUrl = 'data:image/png;base64,' + Base64.fromBase64Url(sample);
> ```

Thank you. I tried `this.fingerprintImgString = 'data:image/png;base64, ' + Base64.fromBase64Url(rawData);` but no luck. 
I copy my result to https://codebeautify.org/base64-to-image-converter to test, the result is the same :(

This is my base64 result:
[base64string.txt](https://github.com/hidglobal/digitalpersona-devices/files/7679379/base64string.txt)

a-bronx, 2021-12-09:
If you decode the data in your `base64string` file into ASCII you'll see it is a JSON encoding a [`BioSample`](https://github.com/hidglobal/digitalpersona-core/blob/20dfa8d959479f5b2992f31b1ff8733d1089ff3e/src/biometrics/factor.ts#L110) structure:

```jsonc
{
  "Compression": 0,
  "Data": "_v7-_v7-_v7-_v7 .......  _________________________8",   // <<< base64url-encoded image data!
  "Format": {
    // ...
  },
  "Header":{
    "DeviceId": 72345365437350147,
    "DeviceType": 6296422711555,
    // ...
  },
  "Version":1
}
```

The `sample.Data` is what you're looking for. It is a **`Base64Url`**-encoded image data, not `Base64`. If you need an image URL, you have to convert it from base64url to base64.
Try the following code (I'll make it more verbose, so you can see every step and all typings):

```ts
function onSamplesAcquired(data: SamplesAcquired)
{
    const samples: Base64UrlString[]          = data?.samples;
    if (!samples || samples.length < 1) return;
    const encodedSampleData: Base64UrlString  = samples[0];
    const decodedSampleData: Utf8String       = Utf8.fromBase64Url(encodedSampleData);
    const sample: BioSample                   = JSON.parse(decodedSampleData);
    const encodedImageData: Base64UrlString   = sample.Data;
    const base64imageData: Base64String       = Base64.fromBase64Url(encodedImageData);
    const imageUrl                            = `data:image/png;base64, ${base64imageData}`;
   ...   
};
```

pdhung197, 2021-12-09:
Hi @a-bronx . Thank you for your support.
I guest you can not test the code, so have some issues in your above example:
`samples[0]` is not a `Base64UrlString`, `samples[0].Data` is `Base64UrlString`.
So, I follow your instruction and this is my code:

```
const { Data: rawData } = samples[0];

const decodedSampleData: Utf8String = Utf8.fromBase64Url(rawData);
const sample: BioSample = JSON.parse(decodedSampleData);
const encodedImageData: Base64UrlString = sample.Data;
const base64ImageData: Base64String =
Base64.fromBase64Url(encodedImageData);

/* domSanitizer just for Angular to make sure this is a safe url */
this.fingerprintImgString = this.domSanitizer.bypassSecurityTrustUrl(
   `data:image/png;base64, ${base64ImageData}`
);
```
But the image still fail, not show.
Do I make some mistake in my code? My `SampleFormat` in `startAcquisition` is `Raw`.

I attach my Base64Url string and Base64 string for testing if you need.
[rawData-base64Url.txt](https://github.com/hidglobal/digitalpersona-devices/files/7681921/rawData-base64Url.txt)
[base64string.txt](https://github.com/hidglobal/digitalpersona-devices/files/7681922/base64string.txt)

a-bronx, 2021-12-09:
Yes, I haven't have a chance to test my code, I was writing it from my head.

What `SampleFormat` do you use?

pdhung197, 2021-12-09:
@a-bronx I'm using Raw. `SampleFormat.Raw`

a-bronx, 2021-12-09:
If you use `data:image/png;base64`, then you must use `SampleFormat.PngImage` correspondingly.

pdhung197, 2021-12-09:
@a-bronx I'm using Raw. For PngImage, I see it's type is BioSample, but get samples[0].Data for Base64UrlString return undefined.

a-bronx, 2021-12-09:
Your URL explicitly says that its MIME type is expected to be `image/png`. You cannot use a raw bitmap and "cast" it to a PNG image, its a totally different format.

Can you attach a content of the `data: SampleAcquired` argument of the event handler?

pdhung197, 2021-12-09:
Hello @a-bronx , I found an issue that the type of samples in data when using SampleFormat.PngImage is BioSample[]. In fact, they are Base64UrlString[].

I'm using this code and it worked:

```
const base64ImageData: Base64String = Base64.fromBase64Url(
  samples[0] as unknown as Base64UrlString
);

this.fingerprintImgString = this.domSanitizer.bypassSecurityTrustUrl(
  `data:image/png;base64, ${base64ImageData}`
);
```

a-bronx, 2021-12-09:
@pdhung197 Thank you for reporting the typing issue!

birladotech, 2022-07-25:
after having the data in base64 or base64url format with the sampleData. I want to be able to save the data to a database and be able to compare the fingerprint of the people again. I use an api to send the data to my server but I can't compare the users.

a-bronx, 2022-07-25:
To compare fingerprints, you need some fingerprint matching engine, like HID DigitalPersona FingerJet (a part of [HID DigitalPersona Suite](https://www.hidglobal.com/products/software/digitalpersona/digitalpersona)).

radhe65gupta, 2023-08-26:
Hey @a-bronx  ,i have old U.are.U 4500 DigitalPersona Fingerprint Reader.This old Reader Work on ASP.Net Core  Web app. Please Help me regarding this issue.

a-bronx, 2023-08-26:
@radhe65gupta I see no issue if it works. Note also, this issue tracker is for a Javascript library, not ASP.NET. If you have an issue with the @digitalpersona/devices library, please open a new issue.

radhe65gupta, 2023-09-04:
> Try:
> 
> ```
> import { Base64 } from '@digitalpersona/core';
> const sampleImageUrl = 'data:image/png;base64,' + Base64.fromBase64Url(sample);
> ```

hey @Abdullah0991  i want to use this Png file to Create Fmd and Compare in Asp.net Core. Please Help me .

Abdullah0991, 2023-09-04:
How Can I help you?

radhe65gupta, 2023-09-04:
> How Can I help you?
i need sample, which create Fmd with the help of Png data and store in DB and verify again finger  data with current Finger png data with capure by SampleFormat.PngImage  SampleAcquired. in ASP.Net Core

radhe65gupta, 2023-09-04:
> after having the data in base64 or base64url format with the sampleData. I want to be able to save the data to a database and be able to compare the fingerprint of the people again. I use an api to send the data to my server but I can't compare the users.

your issue had sloved or not

a-bronx, 2023-09-05:
@radhe65gupta The `@digitalpersona/devices` library will not help you to "_create Fmd with the help of Png data and store in DB and verify again finger data with current Finger png data_". [This library is for fingerprint capture only](https://hidglobal.github.io/digitalpersona-devices/how-to.html). 

For finger verification you need a fingerprint matching engine running on the server, like the DigitalPersona Server, or DigitalPersona U-are-U SDK. These products come with a support and samples. If you like to use the DigitalPersona fingerprint engine, please contact HID for a quote. If you prefer to use a fingerprint engine of any other vendor, you should contact the vendor.

radhe65gupta, 2023-09-05:
> @radhe65gupta The `@digitalpersona/devices` library will not help you to "_create Fmd with the help of Png data and store in DB and verify again finger data with current Finger png data_". [This library is for fingerprint capture only](https://hidglobal.github.io/digitalpersona-devices/how-to.html).
> 
> For finger verification you need a fingerprint matching engine running on the server, like the DigitalPersona Server, or DigitalPersona U-are-U SDK. These products come with a support, guidelines and samples. If you like to use the DigitalPersona fingerprint engine, please [contact HID for a quote](https://www.hidglobal.com/products/digitalpersona). If you prefer to use a fingerprint engine of any other vendor, you should contact the vendor.

Thanks @a-bronx

josephjoberno, 2023-09-05:
How to use digital persona sdk with my fingerprint ong to compare??

a-bronx, 2023-09-05:
@josephjoberno Please, refer to the DigitalPersona U-are-U SDK documentation and samples, or contact the HID customer support. This repository is not a suitable place for issues related with the U-are-U SDK.

radhe65gupta, 2023-09-06:
> […](#)

Thanks @josephjoberno

rangerisrael, 2023-10-19:
sir @a-bronx  it's possible to convert PNG Image to Intermediate sample format or Intermediate to PngImage Format?