hidglobal/digitalpersona-devices #13: React Issue

michaelfemi81, 2021-02-26:
Uncaught Error: Cannot find module 'WebSdk'
I have gone through all the issues and I have tried to implement everything.]
I included Websdk to index.html and still same issues. How hard is it to include one module and everyone keeps getting this issue

a-bronx, 2021-02-26:
The `WebSdk.js` must be included on a **browser side only**. Make sure it is not included in any server-side rendering part. Most of the issues are caused by developers including the WebSdk.js in a wrong place (SSR), not because it is not a separate module.

michaelfemi81, 2021-02-26:
Im using the create-react-app template so it is not server rendered, I had to comment out the import websdk from the file in the node modules folder . But now the device is not discovered and all drivers have been - installed

a-bronx, 2021-02-27:
Add `<script src="path/to/WebSdk/index.js"></script>` [in your HTML page](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/2c54be9c09434bdac39298162e4e6ff7316038c7/src/index.html#L31). Make sure the browser successfully loads the script into the page.

If you need to [import WebSdk](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/2c54be9c09434bdac39298162e4e6ff7316038c7/src/index.ts#L11) for types, make sure you have [WebSdk typings](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk) in the import path. You should not import any library code into your JS/TS, only typings.

michaelfemi81, 2021-02-27:
I already did all that and the issue im getting now is that, the browser doesn't detect the fingerprint

michaelfemi81, 2021-02-27:
I get Uncaught (in promise) Error: Communication failure... When I call this function reader.startAcquisition(SampleFormat.Intermediate);

a-bronx, 2021-02-27:
Do you have DigitalPersona Lite Client installed on the client machine? See [this thread](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/2c54be9c09434bdac39298162e4e6ff7316038c7/src/index.html#L31).

michaelfemi81, 2021-02-27:
Yes I installer the device driver and the fingerprint device is blinking, the blinking changes when I run my thumb over it, so im assuming its working

a-bronx, 2021-02-27:
The Lite Client is not a device driver, it is a user agent providing a local web service and a communication channel between the hardware driver and a browser. Browsers are sandboxed and cannot communicate with devices directly, hence the need of the agent. You should see a `DpAgent.exe` process running in your Task Manager.

See [more details and links in the documentation](https://hidglobal.github.io/digitalpersona-devices/).

michaelfemi81, 2021-03-01:
I tried installing this software and it is not compatible with my Windows 10 computer. Also is it possible to get the scan in FMD format directly from the javascript so I can compare directly with the data stored in the db

a-bronx, 2021-03-01:
There should be no Win10 compatibility problems in the Lite Client. Can you share details here? If you have an installation error, can you provide an MSI log ([turn logging on](https://docs.microsoft.com/en-us/troubleshoot/windows-client/application-management/enable-windows-installer-logging) before installing)? If the Lite Client is installed but still cannot connect, check Windows Event Log for errors.

Also, what browser do you use? Note that for MS Edge browser you may need to enable a loopback exemption (e.g. with a PowerShell command `CheckNetIsolation LoopbackExempt -a -n="Microsoft.MicrosoftEdge_8wekyb3d8bbwe"`), because Edge prevents pages to connect to `localhost` by default.

About formats: the library can provide minutia data in a DigitalPersona proprietary format ([`SampleFormat.Intermediate`](https://github.com/hidglobal/digitalpersona-devices/blob/f5d8e49f5da02f12580d859ac2d2248507209e6e/src/devices/fingerprints/sample.ts#L8)) which can be used with DigitalPersona fingerprint matching engine only. If you need compatibility with other matching engines, you need to use your own feature extractor on raw fingerprint images.

michaelfemi81, 2021-03-01:
Thank you I have figured it out somehow. Also Is the away I can just use the library to match saved data from simpleformat

a-bronx, 2021-03-01:
This library is for fingerprint acquisition only. To use `SampleFormat.Intermediate` for enrollment and authentication you'll need either the DigitalPersona Authentication Server (if you need a turn-key solution integrated with ActiveDirectory or LDS), or the DigitalPersona SDK (if you want to do your own solution using DigitalPersona fingerprint matching engine). For that please inquire HID DigitalPersona [Professional Services](https://www.hidglobal.com/professional-services) team.

michaelfemi81, 2021-03-02:
Thanks I will close the issue now, but the library is not useful for me and I just wasted my money on the devices I got as all I need is just to confirm that a user entered an information, and there is no easier way to do this and there is no time to start creating my own feature extraction and matching engine, it will be very nice if we can just compare samples gotten from the library, that will open doors to alot of applications

closed 2021-03-02 by michaelfemi81

kashanalikhalid, 2021-07-27:
hello michael can you tell me how you solved the issue? i am stuck on same thing
Thanks in advance

omerts, 2021-12-21:
I have no server side rendering and it is still failing with the same error during the build

FelipeLozano100, 2022-01-28:
Lograron solucionarlo?

fayizqureshii, 2023-12-07:
Hey @michaelfemi81 , Do you figure this out with react?

omarasael1980, 2024-02-18:
Could anyone implement it in React? It has only worked for me in Angular, I need to be able to read the fingerprint and extract the base64

omarasael1980, 2024-02-22:
I finally managed to implement fingerprint scanning in my project, displaying it as an image or base64 in the console. You can find the code on my GitHub: https://github.com/omarasael1980/digitalPersonaReact/. It's a simple fingerprint reading implementation using @digitalpersona/devices in React. Feel free to check it out, and I hope it proves helpful to anyone working on a similar project!