hidglobal/digitalpersona-devices #14: Problem with the driver

michaelfemi81, 2021-03-18:
I bought a new device and it comes with a cd which is supposed to contain the drivers, the cd software is corrupt and we cannot install it, for this reason we cant use the library on the web, the url for download sent to me is bad too, after installing the software, the program installed doesn't open. Please this is really bad, It is really hard to come across a working driver or whatever software is used to detect the fingerprint device. This is really delaying a major project

michaelfemi81, 2021-03-18:
All I need is to be able to capture the user fingerprint as png and send to my server, the browser cant even detect the reader cause I cant even find a working driver on the internet, I think a link should be added to the repo, cause the cd is not even working and not every computer has a cd rom

a-bronx, 2021-03-18:
Please note that this repository is for the specific JavaScript library only. The issue tracker here is not for reporting issues about hardware, drivers and unrelated software.

Please use other channels to resolve your issues with the device drivers:
* the party who sold you the device, provided URLs etc.
* the manufacturer of the device and drivers.

closed 2021-03-18 by a-bronx