hidglobal/digitalpersona-devices #17: sampleQuality always return Good!

Abdullah0991, 2021-07-10:
Hello,

Why always the `sampleQuality` returns `Good` in the `onQualityReported` event ? I tried capture a lot of wrong finger placements but always quality was good.

And another problem, when capturing a fingerprint sample with a bad finger placement, the `onSamplesAcquired` returns a corrupted fingerprint sample (using it in `img` shows corrupted image). Is there a possibility to avoid acquiring these samples?

FYI: Using the lib in Angular 11 app.

a-bronx, 2021-07-11:
Hello,

I'll check with the fingerprint API developers after weekends, but my guess is that the DigitalPersona Fingerprint API (the underlying native component this library uses, part of DigitalPersona Workstation or Lite Client) reports image quality only when it extracts features, transforming it to fingerprint templates. This requires a fingerprint engine installed on the client machine, which is not the case for the DigitalPersona Lite Client. So the fingerprint image is returned "as-is", without quality analysis, and it is expected that the fingerprint engine of your choice will perform the analysis during feature extraction,

a-bronx, 2021-07-13:
You still may receive a bad quality status in cases detected by a fingerprint device driver, such as a extremely low image contrast (e.g. when your finger's surface is overly dry), wrong swipe direction (skewed or inverted) etc. Device drivers have limited capabilities, so they detect only basic scan flaws. They may pass images with smudged ridges, wrong placement etc as "good" ones. This depends on a specific device model and its driver.

closed 2021-07-13 by a-bronx