hidglobal/digitalpersona-devices #21: Can not test failed case and convert raw data image to image src

pdhung197, 2021-12-08:
Dear HID,

First, I'm testing event of fingerprint reader, but I can not test the "failed" case.
In QualityReport, I alway receive data like this:

```
deviceId: "8C630117-B87E-6B47-8D2E-5AAF6EAACC9A"
quality: 0
type: "QualityReported"
```
.... even when I intentionally put my whole hand on chin on the device. Another cases can not test because no event's called.
So, how can I test the failed case and got correct quality (put whole the hand or chin on the device can not is a good quality case).

The second. I use this solution to convert Sample raw data to image src but seem like no luck, or have something wrong in my code.
This is my code:
In SampleAcquired event:

```
const { Data: rawData } = samples[0];
const sampleImageUrl = 'data:image/png;base64,' + btoa(Utf8.fromBase64Url(rawData));
this.imgString = sampleImageUrl;
```
In template: `<img [src]="imgString" alt="" />`
And the result like this:

[image]

Is there any mistake from me that make the code not work? :(

pdhung197, 2021-12-09:
1. Duplicate with [#17](https://github.com/hidglobal/digitalpersona-devices/issues/17)
2. Solved in [#11](https://github.com/hidglobal/digitalpersona-devices/issues/11#issuecomment-989682746)

Close issue.

a-bronx, 2021-12-09:
Thanks for reporting.

Typing for the `SamplesAcquired` needs to be fixed, and more detailed documentation needed for different types of sample formats. Will keep the issue open until that.

a-bronx, 2021-12-09:
Some findings for reference:

* for the `SampleFormat.Raw`, the `BioSample.Data` member appears to be a base64url-encoded octet-stream of a bitmap content, while bitmap metadata is encoded in a missing `BioSample.Format` member (missing in `BioSample` typing):
```jsonc
{
  "Compression": 0                  //<< missing in BioSample typing
  "Data": "........",
  "Format": {                       //<< missing in BioSample typing
    "iHeight": 550,
    "iWidth": 500,
    "iXdpi": 700,
    "iYdpi": 700,
    "uBPP": 8,
    "uDataType": 1,
    "uImageType": 2,
    "uPadding": 2,
    "uPlanes": 1,
    "uPolarity": 2,
    "uRGBcolorRepresentation": 0,
    "uSignificantBpp": 8
  },
  "Header": {                       //<< does not match BioSampleHeader typing
    "DeviceId": 72345365437350147,
    "DeviceType": 6296422711555,
    "iDataAcquisitionProgress": 100,
    "uDataType":1
  },
  "Version":1
}
```

* For the `SampleFormat.PngImage`, it is reported that `SamplesAcquired.samples` member is not `BioSample[]` but `Base64UrlString[]`, where each element of the array is a base64url-encoded PNG image.