hidglobal/digitalpersona-devices #22: Bad request on initialization

ultimaterichy, 2022-01-30:
Hello, a
[image]
ll of a sudden I started getting an error from javascript and device does not get detected. Few days ago the code was working fine.
I tried running the sample from digitalpersona repo but still got same error(sample used to work when I first ran it two months ago).

`POST https://127.0.0.1:9001/connect 400 (Bad Request)     index.js:5957 `

I tried https://127.0.0.1:52181/get_connection and got expected response. What could be the problem?

I tried the following sdk versions: 2.2.3 and 3.2.0 and I also tried the client lite version

a-bronx, 2022-01-31:
Can it be that the port 9001 is blocked by a firewall or taken by some other service?

ultimaterichy, 2022-01-31:
> Can it be that the port 9001 is blocked by a firewall or taken by some other service?

The port 9001 is not blocked and it is used by DPAgent service.

At some point, I restarted my PC and it worked then I restarted the PC again and it stopped working

a-bronx, 2022-02-02:
Sounds like an external issue not related with the library itself. Try to temporarily disable AV or firewall, clear the browser cache, check for system/app errors in Event Viewer. Try to kill DpAgent.exe process and then restart it manually (from %ProgramFiles%\DigitalPersona\Bin) without rebooting.

ultimaterichy, 2022-02-03:
After doing as suggested in the comments, I got it working but same problem occur after every reboot. I decided to switch to using firefox and the problem never occurred(after properly configuring firewall) even after rebooting several times.

a-bronx, 2022-02-04:
Can you share a name and a version of the offending browser, and a version of the OS?

closed 2023-07-14 by a-bronx