hidglobal/digitalpersona-devices #26: Uncaught ReferenceError: assignment to undeclared variable i

Isfan14, 2022-05-29:
https://github.com/hidglobal/digitalpersona-devices/blob/fcbeaaddca28d6651a2c12b6656af9c8860de925/%40types/WebSdk/index.js#L4677

I got this error when trying to use it in my Nuxt3 project. How to fix it?

a-bronx, 2022-05-29:
Thank you, I'll pass the issue to the developer of the WebSdk dependency and update the sample,

locent217, 2024-01-09:
I have the same issue and I can't use it on my vue app
[image]

a-bronx, 2024-01-09:
Please try the updated WebSdk module [from here](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src/modules/WebSdk), this should solve the issue with a local var declaration.

closed 2024-01-09 by a-bronx

locent217, 2024-01-14:
@a-bronx I copied the index.js code from the link you provided, and still the i = 0; does not have var declaration. so i just manually updated on my local file. thank you!

a-bronx, 2024-01-16:
@locant271 can you tell the line number where the `i = 0;` without `var` happens?

locent217, 2024-01-16:
@a-bronx There's a lot of it. I Just searched all lines with "(i = 0" and changed it to (var i = 0

a-bronx, 2024-01-16:
@locent217 But which of them causes an issue? Every remaining instance follow a common (and benign) pattern: 
```js
function foo() {
    var i;                          // << creates a variable in a function-level scope
    ...
    for(i = 0; ,,,) {}            // << uses the local variable
}
```
Unlike the initial report, this pattern should not cause any issues as each `var` creates a function-level scope and do not mess with globals. So there is no real need "fix" what is not broken.

locent217, 2024-01-16:
@a-bronx Hello! From the lines 1303, 1318, and 1329. Can you check if it does cause the error?

locent217, 2024-01-16:
@a-bronx my bad, I just realize that the link provided from the tutorial leads to another branch, not the master branch, that is why I am getting the error. I fixed it now by copying the code from master branch. thanks!

a-bronx, 2024-01-16:
@locent217 Oh, good catch, thank you! I fixed links.