hidglobal/digitalpersona-devices #27: Can't resolve 'WebSdk' (React, Windows 10, U.are.U 4500 Fingerprint Reader)

ahmed-alii, 2022-06-23:
_H, Please take a look at the following. I have read the issues specifically #4 but not sure how it can be resolved in react app_

`./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js
Module not found: Can't resolve 'WebSdk' in 'E:\digitalpersona-react-sample\node_modules\@digitalpersona\devices\dist\es5.bundles'
`

I am experiencing the above issue on a react sample provided [here](https://github.com/folajubril/digitalpersona-react-sample)

**My Setup**
- Drivers Installed in Windows 10
- Lite Client Installed
- Device listening on port 52181 & port 9001 is also open
- `https://127.0.0.1:52181/get_connection ----->>>  {"endpoint":"https://127.0.0.1:9001/?web_sdk_id=5e5a1211-0b1d-4f3a-8b18-ee057f7a6127&web_sdk_minport=9001&web_sdk_port=9001&web_sdk_secure=true&web_sdk_username=kY_fyw4wmJ4&web_sdk_password=8D2E5C87BDBE072F&web_sdk_salt=ACDCB7545A8C39CB3BB93613FEA54C9377A3A62C6ED6B9F545C49DF014DA13D3"}`

This react app is rendering Client Side. So any kind of library included runs on the browser. 

Index.html contains the  `websdk.client.min.js` file. 

Other Libraries are included inside App.js
`import {BioSample} from "@digitalpersona/core"; `
`import {FingerprintsAuth} from '@digitalpersona/authentication'; `
`import {IAuthService} from '@digitalpersona/services'; `
`import {FingerprintReader, SampleFormat} from '@digitalpersona/devices'; 
`

a-bronx, 2022-07-19:
Hello,

The WebSdk is a browser-only library, and must be loaded **only** into a browser context -- either as a static resource, or as a part of a bundle. The error message suggests that you try to load the WebSdk from the `node_modules`, which is clearly not expected to be seen in a browser. You must exclude the WedSdk library from a Node module resolution, and the code that depends on it must execute only in a browser as well.

closed 2022-07-19 by a-bronx

teknosains, 2023-02-13:
> Hello,
>
> The WebSdk is a browser-only library, and must be loaded **only** into a browser context -- either as a static resource, or as a part of a bundle. The error message suggests that you try to [...]

no example provided, im frustated :(. Can you please help how to use this in vanillaJS or React

a-bronx, 2023-02-14:
Hello, the link to an [example](https://github.com/hidglobal/digitalpersona-sample-angularjs/tree/master/src) was provided many times, in a [documentation](https://hidglobal.github.io/digitalpersona-devices/index.html) and in a pinned issue #4.

Salman9000, 2023-02-26:
I managed to solve it. Make sure your websdk is in src folder.
Inside your node_modules, go to @digitalpersona/devices/dist/es5.bundles/index.umd.js
change line number 2 from `typeof exports === 'object' && typeof module !== 'undefined' ? factory(exports, require('@digitalpersona/core'), require('WebSdk')) :`
to `typeof exports === 'object' && typeof module !== 'undefined' ? factory(exports, require('@digitalpersona/core'), require('../../../../../src/webSdk')) :`

done :)

a-bronx, 2023-02-27:
> `require('../../../../../src/webSdk'))`

While it may work in your specific case, I cannot recommend it. If you need to resort to hard-coded import paths instead of a module name, then it is most probably something wrong with your module resolution settings. I'd advise to find and fix the root cause of the problem. The WebSdk library must be in a list of "globals", like any other global JS library. For example, if you use a Webpack, then you need to add in into your Webpack config like this:
 
```
module.exports = {
   ...
    externals: [
        { WebSdk: {
            root: "WebSdk"
        }},
       // ... other globals  
   }
  ...
}
```
Also, if you use Typescript, you may need to add the path to your global modules into the `tsconfig.json`:

```
      "typeRoots": [
        "node_modules/@types",
        "./@types",
        "src/modules"
      ],
```
-- given that the WebSdk's `index.js` and `index.d.ts` are placed in the `src/modules/WebSdk/` directory

Check documentation for your bundler and module resolution algoritm.

Salman9000, 2023-03-01:
I am using CRA, and the built-in webpack configuration is a spaghetti for me. It is better if i dont touch the webpack

locent217, 2024-01-10:
@a-bronx Hello sir! Can you help me on how to add WebSdk on globals of vite? Thank you!

a-bronx, 2024-01-10:
I have no familiarity with Vite, you'll need to read their docs.

cyheum, 2024-09-20:
@CZEJAY 

I finally solved this problem in Next.js

1. Comment out import 'Web Sdk' ​​in the 3rd line of
 node_modules/@digitalpersona/devices/dist/es5/devices/websdk/channel.js

2. Add Websdk to node_modules.
[image]

3. Wrap components using dynamic
```js
const DynamicComponent = dynamic(
  () => import('./layout/Component').then((mod) => mod.SomeComponent),
  { ssr: false }
)

export { DynamicComponent }

// app/page.tsx

import { DynamicComponent } from 'your component location'

const Page = () => {
  return <DynamicComponent />
}

export default Page
```

I spent all day solving this and I'm so glad I did.

I hope other next js users can solve this problem too. @a-bronx

a-bronx, 2024-09-20:
Again, I'm not familiar with every JS build platforms and their quirks and, especially, someone's specific project configuration permutation, so it would help drasically if a bug report includes a link to a repo with a minimal project with their specific configuration reproducing the problem.

Do I understand it correctly that Next.js and Vite do not support import of vanilla global UMD modules from a folder outside `node_modules`?

a-bronx, 2024-09-20:
@CZEJAY, @cyheum : Aren't you supposed to use the [`next/script`](https://nextjs.org/docs/pages/building-your-application/optimizing/scripts) instead to load external scripts?

EmanuelA20, 2024-09-22:
> @CZEJAY
> 
> I finally solved this problem in Next.js
> 
> 1. Comment out import 'Web Sdk' ​​in the 3rd line of
>    node_modules/@digitalpersona/devices/dist/es5/devices/websdk/channel.js
> 2. Add Websdk to node_modules.
> 
> <img alt="스크린샷 2024-09-20 오후 10 45 04" width="173" src="https://private-user-images.githubusercontent.com/61037219/369428582-d3edbc42-3bb7-4637-9830-8da3d396c916.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwMzI0OTYsIm5iZiI6MTcyNzAzMjE5NiwicGF0aCI6Ii82MTAzNzIxOS8zNjk0Mjg1ODItZDNlZGJjNDItM2JiNy00NjM3LTk4MzAtOGRhM2QzOTZjOTE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIyVDE5MDk1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwNTc0Njk1YzdiNTQzMDgzNTUzZjdhNTgwOWU4NDQzZjAxNDA0M2JjZjVhODFmZjhhZWRkNGQ1MDNlNDYwYzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.j9W04XBfS_KcL-J5nVMnXhQi4qhJfsCDqWtXzJl-HmI">
> 3. Wrap components using dynamic
> 
> ```js
> const DynamicComponent = dynamic(
>   () => import('./layout/Component').then((mod) => mod.SomeComponent),
>   { ssr: false }
> )
> 
> export { DynamicComponent }
> 
> // app/page.tsx
> 
> import { DynamicComponent } from 'your component location'
> 
> const Page = () => {
>   return <DynamicComponent />
> }
> 
> export default Page
> ```
> 
> I spent all day solving this and I'm so glad I did.
> 
> I hope other next js users can solve this problem too. @a-bronx

Hello, Were you able to connect the Digital Persona 4500 fingerprint reader with the Web SDK? If so, I would like to know if you have any reference, because I have not been able to. I see that they say that in order to connect it to the web you have to use Java, C#, etc. I would like to find something feasible with JS to capture and compare fingerprints. I would appreciate your help.

cyheum, 2024-09-23:
@a-bronx
In my opinion, it is possible to load an external library through the next/script or script tag in next.js, but the problem seems to be that the library in node_modules relies on an external library loaded through an internal script. It's not accurate, so I think I need to do more research.

cyheum, 2024-09-23:
@EmanuelA20 
If you use React, you can refer to this repo: https://github.com/omarasael1980/digitalPersonaReact

EmanuelA20, 2024-09-23:
> @EmanuelA20 If you use React, you can refer to this repo: https://github.com/omarasael1980/digitalPersonaReact

@cyheum Wow, after so much time, with that repository I managed to capture the fingerprint template to save in the db. The problem now is comparing the fingerprints, sorry for the inconvenience, and thank you for the help so far, do you know how I can compare those fingerprints? Is there no method? I use react and nodeJS

a-bronx, 2024-09-23:
@cyheum, well, a library published in NPM relying on a library not published in NPM is nothing unusual, there are many useful libraries which are not published in NPM by their authors for different reasons, but most bundlers support this scenario just fine. Unfortunately, every JS tool does it in its own unique way, and library authors just cannot keep up with every new toolset which are popping up like mushrooms after rain, so it is up to you to learn your toolset of choice.

a-bronx, 2024-09-23:
@EmanuelA20 about fingerprint matching, see the last 4 answers in [the F.A.Q.](https://hidglobal.github.io/digitalpersona-devices/how-to.html)