hidglobal/digitalpersona-devices #33: WebSdk not found

Bartik-ing-programmer, 2022-09-12:
[I have the same problem](https://github.com/hidglobal/digitalpersona-devices/issues/4)

I have tried to solve with the suggestions that they show in that issue, and none of them gives me the solution.

I am currently working on an app that integrates React and Electron

[image]

---

 #### index.html

[image]

Bartik-ing-programmer, 2022-09-12:
I have this error in the execution the script in `index.html`

```bash
Blocked script execution in 'data:text/html' ...(path index.html)...
because the document's frame is sandboxed and the 'allow-scripts' permission is not set.
```

teknosains, 2023-02-13:
I have same issue here, look likes the @digitalpersona modules incluse WebSdk but its not there. in the tutorial, WebSdkl should be loading as a reguler JS modules, not a nodejs module...well its frustating

a-bronx, 2023-02-14:
Hello!

Are you trying to load the library into NodeJS like a regular node module? It won't work, both the library and WebSdk are browser-only and must be loaded into a browser, together.

Also note, we do not guarantee the library working in Electron, never tested it. To works with native drivers of fingerprint readers the library uses a connection from a browser context to the loopback (`localhost`) endpoint provided by the DigitalPersona Lite client. Electron may or may not block this.

The blocked script execution is a probably a CORS issue due to violation of a same-origin policy, no a library issue.

closed 2023-07-14 by a-bronx

franzuandrez, 2025-03-04:
Hey, any updates?