hidglobal/digitalpersona-devices #35: Device Id

crissulb95, 2022-12-02:
hello again.

Im trying to get a **_unique id_**  from my fingerprint reader, but anytime i disconnect and connect my device on any pc and try to get it to log into the console thru getDeviceInfo(deviceId) method, is never the same.

Is it like this on purpose or is something else happening?

a-bronx, 2022-12-03:
Hello!

A word from a WebSdk developer:

> Not all fingerprint devices are having unique `deviceId`. Different devices from different vendors have different approaches and most of the devices have no any unique Id associated with it. From our point of view if device does not have any persistent unique device Id we would randomly generate it when device gets connected to USB. To know if device has persistent or volatile you would need call `getDeviceInfo()` and check `DeviceInfo.eUidType` field of a `DeviceUidType` type. The `DeviceUidType` enumeration lists types of unique identifiers of fingerprint readers:

    enum DeviceUidType {
        Persistent = 0,
        Volatile = 1,    
    }

> So if `eUidType` set to 0, this means the `deviceId` is persistent and would not change after device was disconnected from USB and connected again. If `eUidType` set to 1, this means the `deviceId` is volatile and would change every time after device was disconnected from USB and connected again.

closed 2022-12-03 by a-bronx

crissulb95, 2022-12-07:
Thanks for your quickly response!!