hidglobal/digitalpersona-devices #37: Communication failure: Digital Persona 4500

okothhillary, 2023-03-15:
I keep getting a communication error when I try to connect DP 4500 to my web app. Below is a test file that reproduces the issue I'm facing. I already have DigitalPersona Lite Client and DIGITALPERSONA 4500 WBF DRIVER installed.

[test.zip](https://github.com/hidglobal/digitalpersona-devices/files/10977205/test.zip)

okothhillary, 2023-03-15:
That error is displayed on all browsers that I've tried. I've allowed loopback in MS Edge, I've checked if there's any firewall blocking connection but I haven't seen any after posting "https://127.0.0.1:52181/get_connection", I got something like "endpoint...web_sdk". How do I resolve the communication error?

[image]
[image]

okothhillary, 2023-03-15:
All local agents are running...how do I solve the communication error @a-bronx, @daniilr, @adricasti, @pboulvais ?

DanielIcc, 2023-05-24:
same error here any sugestions?

a-bronx, 2023-07-14:
Apologies for a long delay, I was prioritized to work on different projects.

Upon a code inspection I can only say, that the `this.reader.channel.webChannel.connect()` call in the `FingerprintSigninControl` constructor is unnecessary, the `FingerprintReader.startAcquisition()` will try to open the channel automatically in a proper way.

Note that the company policy forbids me to run your test sample with untrusted content. Should you wish me to investigate further, please create a minimal GitHub project reproducing the issue (no third-party libraries). Also specify versions of the browsers you tried and the version of the Lite Client installed.

closed 2024-01-09 by a-bronx