hidglobal/digitalpersona-devices #43: Xamarin.Android Project | Enrollment | Error on Capture

jlinker7, 2023-07-14:
I am integrating the fingerprint reader into a Xamarin.Android project (Android 7.1).  I used a bindings project with the jar and the .so files.  It works for general scanning, but when I attempt to use the enrollment process when I call the readers Capture method I get the following error.

	java.lang.UnsatisfiedLinkError: No implementation found for int com.digitalpersona.uareu.jni.Dpfpdd.DpfpddCaptureExt(
	long, int, int, int, int, int, com.digitalpersona.uareu.jni.Dpfpdd$IntReference, com.digitalpersona.uareu.jni.Dpfpdd$IntReference, com.digitalpersona.uareu.jni.Dpfpdd$IntReference, com.digitalpersona.uareu.jni.Dpfpdd$IntReference, com.digitalpersona.uareu.dpfpdd.FidImpl) 
	(tried Java_com_digitalpersona_uareu_jni_Dpfpdd_DpfpddCaptureExt and Java_com_digitalpersona_uareu_jni_Dpfpdd_DpfpddCaptureExt__JIIIIILcom_digitalpersona_uareu_jni_Dpfpdd_00024IntReference_2Lcom_digitalpersona_uareu_jni_Dpfpdd_00024IntReference_2Lcom_digitalpersona_uareu_jni_Dpfpdd_00024IntReference_2Lcom_digitalpersona_uareu_jni_Dpfpdd_00024IntReference_2Lcom_digitalpersona_uareu_dpfpdd_FidImpl_2)
	at com.digitalpersona.uareu.jni.Dpfpdd.DpfpddCaptureExt(Native Method)
	at com.digitalpersona.uareu.jni.Dpfpdd.capture(Dpfpdd.java:183)
	at com.digitalpersona.uareu.dpfpdd.ReaderCollectionImpl$ReaderImpl.Capture(ReaderCollectionImpl.java:167)
	at crc64696f8426591dd164.FingerService.n_GetFmd(Native Method)
	at crc64696f8426591dd164.FingerService.GetFmd(FingerService.java:31)
	at com.digitalpersona.uareu.dpfj.EngineImpl.CreateEnrollmentFmd(EngineImpl.java:58)
	
Can you provide any insight into what I am missing?  I attempted using the Metadata.xml to transform the method, but am new to that so may have had it wrong if that is what is required (would need a specific example of what to put in there).  Thanks!

a-bronx, 2023-07-14:
This repository is for a JavaScript library, not Java SDK.

closed 2023-07-14 by a-bronx

okothhillary, 2023-07-14:
seriously...I submitted issue 37 months ago and you never gave me an update on it.

jlinker7, 2023-07-14:
@a-bronx wow...  no help, no here is where you should go, nothing?  Man...

okothhillary, 2023-07-14:
Yeah, lousiest service in the world.

a-bronx, 2023-07-14:
@jlinker7 We have no any online Java SDK repo or issue tracker, I cannot provide you a link, sorry. You should contact whoever you obtained the Java SDK from, for example [HID Global customer support](https://www.hidglobal.com/contact).

This repository is not intended to be a general customer support line.

jlinker7, 2023-07-15:
@a-bronx Thank you, that was much more helpful and informative.  My apologies for my smart remark and I do appreciate the follow up.