hidglobal/digitalpersona-devices #44: WebSdk is not defined Next.js

Dulco, 2023-07-17:
Hi everyone, I got a problem with the WebSdk module using the library @DigitalPersona/devices in next.js.

My problem is the following:

**- error node_modules\@digitalpersona\devices\dist\es5.bundles\index.umd.js (211:38) @ WebSdk
- error ReferenceError: WebSdk is not defined**

I installed the library via "npm", and added the WebSdk Manually to the project. copying the files and creating a folder, to solve module not found I used the webpack configuration to add the module WebSdk to the list. as the following:

**const path = require('path');

module.exports = {
  webpack: (config, { isServer }) => {
    config.resolve.alias['WebSdk'] = path.resolve(__dirname, './public/modules/WebSdk');
    return config;
  },
  async rewrites() {
    return [
      {
        source: '/modules/WebSdk/:path*',
        destination: '/public/modules/WebSdk/:path*',
      },
    ];
  },
};**

, Also I load the module to the browser 
<Script src='/modules/WebSdk/index.js'></Script>

If I go to the node_modules\@digitalpersona\devices\dist\es5.bundles\index.umd.js and check the line where the proble is suppose to be the problem seems to be here:

var Channel = /** @class */ (function () {
        function Channel(channelName, options) {
            if( typeof window != undefined){
                this.pending = [];
                **this.webChannel = new WebSdk.WebChannelClient(channelName, options);**
                this.webChannel.onConnectionSucceed = this.onConnectionSucceed.bind(this);
                this.webChannel.onConnectionFailed = this.onConnectionFailed.bind(this);
                this.webChannel.onDataReceivedTxt = this.onDataReceivedTxt.bind(this);
            }
        }
 
 When creates a new Web Channel, however if I ctrl click the WebSdk, it sends me to the module, so its loaded not undefined.
 
 If anyone could help me with this issue, I've been strugglin for a few days :(

a-bronx, 2023-07-17:
Duplicate of multiple closed issues. Please, do a minimal research before posting the same issue multiple times.

closed 2023-07-17 by a-bronx

Dulco, 2023-07-18:
I don't think it's the same problem as mentioned in any the issues before, but thanks anyway.

a-bronx, 2023-07-18:
Check #36.