hidglobal/digitalpersona-devices #46: Old DigitalPersona U.are.U 4500 Fingerprint Reader

radhe65gupta, 2023-08-28:
I have Old DigitalPersona U.are.U 4500 Fingerprint Reader. How can i integrate  in ASP.Net Core Web App? It cant  integrate  or not?

closed 2023-08-29 by a-bronx

radhe65gupta, 2023-08-29:
Thanks @a-bronx for replying. I'm able to collect finger print base64url encoded  data , and I want to save the finger print data in my server and next time I used this data for verification with c# on server side. It work on .net window application. But I can't achieve this web app.

a-bronx, 2023-08-29:
This library only for collecting fingerprints. Regarding fingerprint matching, see answers [here](https://github.com/hidglobal/digitalpersona-devices/issues/6#issuecomment-668815606) and [here](https://github.com/hidglobal/digitalpersona-devices/issues/32#issuecomment-1204495983).

radhe65gupta, 2023-09-04:
Thanks @a-bronx ,But i don't know ,why @Digitalpersona not provide any developer guidelines regaring this problem? If you have some developer guidelines please help me regarding this issue.

a-bronx, 2023-09-05:
@radhe65gupta The `@digitalpersona/devices` library will not help you to "_create Fmd with the help of Png data and store in DB and verify again finger data with current Finger png data_". [This library is for fingerprint capture only](https://hidglobal.github.io/digitalpersona-devices/how-to.html). 

For finger verification you need a fingerprint matching engine running on the server, like the DigitalPersona Server, or DigitalPersona U-are-U SDK. These products come with a support, guidelines and samples. If you like to use the DigitalPersona fingerprint engine, please contact HID for a quote. If you prefer to use a fingerprint engine of any other vendor, you should contact the vendor.