hidglobal/digitalpersona-devices #47: Could it work with Javascript and on a cloud domain?

javico-dev11, 2023-09-21:
Good afternoon, could this library be implemented in pure Javascript and work in a cloud domain?

a-bronx, 2023-09-21:
Hello,

This library *is* a Javascript library.

This library is a client-side library, it must work in a web browser, and it relies on the DigitalPersona Lite Client running on the same machine, a native Windows application which provides a connection between the browser and a fingerprint reader, which is a physical device. I'm not sure what exactly do you mean by "work in a cloud domain" in this regard. Can you clarify?

a-bronx, 2023-09-22:
The application may be _hosted_ in any domain, using any backend framework, but in the end it must be executed _in a browser_, on the webpage opened on the same machine where the reader is connected to. Thus 'localhost'. 

How would you expect the script to find the reader, if it is running on a completely different machine? The only way to connect the reader to a remote machine is to create a USB "tunnel" in your RDP connection (or in a virtual machine console like VirtualBox or VMWare). The machine where te script is running must see the reader as if it was physically connected to it.

javico-dev11, 2023-09-22:
Yes, exactly, just as you mention, on the PC where I open the web application that executes the script, the agent is installed right there and the reading computer is connected.
My question is rather because it works only when I am testing on localhost, and it does not work when the application is already deployed in the cloud.

okothhillary, 2023-09-22:
Any python implementation for Digital Persona 4500 SDK? After buying one, I
don't see any.

javico-dev11, 2023-09-22:
> Any python implementation for Digital Persona 4500 SDK? After buying one, I don't see any.
> […](#)

You don't see documentation or examples for Python recently!.

okothhillary, 2023-09-22:
Zero

javico-dev11, 2023-09-22:
@okothhillary You don't see documentation or examples for Python recently!, You can also use the C# implementation, if it is for desktop applications.

okothhillary, 2023-09-22:
No, I do not see any.
It's a web app

javico-dev11, 2023-09-22:
@okothhillary  However, if it is a web application, you can implement the JavaScript script in your frontend, it does not matter if the backend is Python.

okothhillary, 2023-09-22:
What about my CRM though? I'm currently using C++ and CMake to now build it
cause I can't find the python implementation

okothhillary, 2023-09-22:
Frontend is fine. Device connects.

javico-dev11, 2023-09-22:
@okothhillary Oh I understand, maybe you have checked this article: [https://jomutech.com/raspberrypibiometricintegration/](url)

okothhillary, 2023-09-22:
I see. But I have to get a raspberry pi...

javico-dev11, 2023-09-22:
@okothhillary Of course, it is for RaspberryPi, I hope the administrator can give you a more specific answer.

okothhillary, 2023-09-22:
I had hoped there's a backend implementation in python explained in their
SDK documentation.

javico-dev11, 2023-09-22:
Have you had the same problem that I am currently having?

a-bronx, 2023-09-22:
> I get an error: not found https://domain.com/get_connection [image]

Did you modify the WebSdk anyhow? Asking because the `localhost` is literally hardcoded into the WebSdk code:

```js
Configurator.prototype.ensureLoaded = function (callback) {
    ...    
    var self = this,
          uri = "https://127.0.0.1:52181/get_connection";

    utils.ajax('get', uri) ....
};
```

How could the initiation URL could change to `https://domain.com/get_connection`? Are there are any other code in your app that tries to make a request to this URL?

a-bronx, 2023-09-22:
@okothhillary 
1. No, there is no Python SDK, but there is a native C SDK which should be pretty easily invokable from Python.
2. If you have a question which is different from the one which opens the thread, I'd suggest you to create a new issue instead of hijacking threads started by others. In future, I will be deleting your comments if they are off-topic. Thank you.

javico-dev11, 2023-09-22:
@a-bronx Thank you very much for your kind response, I have not modified the WebSDK code, but I did see the part where it refers to 127.0.0.1:52181/get_connection, can I edit this without problems?

a-bronx, 2023-09-22:
@javico-dev11 You should not modify this code, it is intended to connect to the localhost. The WebSdk code is also must be executed in the browser, on the machine where thre LiteClient is running and fingerprint reader is connected to.

Can you trace where the call to the `https://domain.com/get_connection` originates from?

javico-dev11, 2023-09-22:
Ok, I understand, that is precisely why I did not modify, and about tracking where the call comes from: [,](https://domain.com/get_connection) it is when I press a button that calls the following function:
`async startFingerprintAcquisition() {
                try {
                    this.timeOut = true;

                    const devices = await this.reader.enumerateDevices();

                    if (devices.length === 0) {
                        throw new Error("No fingerprint reader is connected!");
                    }

                    console.log("count devices: " + devices.length);
                    const deviceUid = devices[0]; 
                    this.deviceUid = deviceUid;
                    console.log("Starting fingerprint acquisition!");
                    let resp = await this.reader.startAcquisition(
                        dp.devices.SampleFormat.PngImage,
                        deviceUid
                    );

                  
                    await this.wait(5);
                    this.timeOut = false;
                    console.log("Stopping fingerprint acquisition");
                    await this.reader.stopAcquisition(deviceUid);
                } catch (error) {
                    console.error('Error en startFingerprintAcquisition: ' + error);
                }
            }`

a-bronx, 2023-09-22:
Can it be that the 'localhost' on your local machine somehow gets translated to the `demo-dw.azurewebsites.net` cloud host address? What happens if you enter the `https://127.0.0.1:52181/get_connection` URL in the browser address line?

javico-dev11, 2023-09-22:
By entering the address as you suggest if it detects the route:
[image]

Could it be that localhost is modified on my machine?

a-bronx, 2023-09-22:
The response looks correct. Do you run your app in the very same browser?

javico-dev11, 2023-09-22:
Yes, I have tried in Chrome, Edge, Firefox and Brave

a-bronx, 2023-09-22:
Is it the `const devices = await this.reader.enumerateDevices();` line that cause the error?

Can you trace it down to the `Configurator.ensureLoaded` call and further down to the `XMLHttpRequest.open(method, url,...)` and verify the `url` is the localhost one but it somehow gets translated to a different host?

javico-dev11, 2023-09-22:
Ok, I'll do it and let you know.
Thank you very much for your support!

javico-dev11, 2023-09-22:
Hello again Andrey, I have re-uploaded the version of my application and tested again in all web browsers. It has worked correctly for me except in Brave, I suppose it is due to some specific browser configuration.

Do you think there was some problem in the deployment of the application and that is why it did not work?

a-bronx, 2023-09-22:
Hard to say. Did you use any VPN or remoting software that could mess with address translation?

What problems do you experience in Brave? Is it related with Brave Shield raised up?

javico-dev11, 2023-09-22:
The error that was showing in all browsers, that is, the one mentioned above, only occurs in Brave, if I have a VPN but it is currently disabled.

[image]

In Firefox or Chrome, it works perfectly, I think I'll leave it there for now so I can continue with work.

a-bronx, 2023-09-22:
Ok, thanks for reporting, I'm glad the issue is mostly resolved.

closed 2023-09-22 by a-bronx