hidglobal/digitalpersona-devices #48: WebSdk and variable issues.

shinomikko8888, 2023-09-24:
Hello! This is my first time using the dependency (and asking for help, in general) and I stumbled into some issues that I cannot resolve that well even when looking at the other issues from #4, #13, #27, and #33. I'm using React.js and most of the examples that were given used Angular, which for me a not-so-experienced dev, cannot find a way to fix the issues that pop-up.

My biometric device is a DigitalPersona U.are.U 4500 Fingerprint Reader and I downloaded the SDK along with the Local Agent from the tutorial page in the documentation. I followed the documentation yet I came up with these errors.
The things that popped up an error were:
1. WebSdk is not defined
```
WebSdk is not defined
ReferenceError: WebSdk is not defined
    at new Channel (http://localhost:3000/static/js/bundle.js:26526:29)
    at new FingerprintReader (http://localhost:3000/static/js/bundle.js:27010:23)
    at FingerprintModal (http://localhost:3000/main.f679d5df578319e64c29.hot-update.js:43:29)
    at renderWithHooks (http://localhost:3000/static/js/bundle.js:66757:22)
    at mountIndeterminateComponent (http://localhost:3000/static/js/bundle.js:70043:17)
    at beginWork (http://localhost:3000/static/js/bundle.js:71339:20)
    at HTMLUnknownElement.callCallback (http://localhost:3000/static/js/bundle.js:56349:18)
    at Object.invokeGuardedCallbackDev (http://localhost:3000/static/js/bundle.js:56393:20)
    at invokeGuardedCallback (http://localhost:3000/static/js/bundle.js:56450:35)
    at beginWork$1 (http://localhost:3000/static/js/bundle.js:76324:11)
```
2. Can't resolve WebSdk
```
ERROR in ./node_modules/@digitalpersona/devices/dist/es5.bundles/index.umd.js 2:115-132
Module not found: Error: Can't resolve 'WebSdk' in 'E:\xampp\htdocs\project\frontend\node_modules\@digitalpersona\devices\dist\es5.bundles'
```

3. Various variable errors
```
ERROR in [eslint]
src\modules\WebSdk\index.js
  Line 28:23:     Unexpected use of 'self'     no-restricted-globals
  Line 28:44:     Unexpected use of 'self'     no-restricted-globals
  Line 28:58:     Unexpected use of 'self'     no-restricted-globals
  Line 28:66:     Unexpected use of 'self'     no-restricted-globals
  Line 2955:40:   'DV' is not defined          no-undef
  Line 4227:36:   'sha1' is not defined        no-undef
  Line 4227:42:   'sjcl' is not defined        no-undef
  Line 4227:48:   'BigInteger' is not defined  no-undef
  Line 4588:21:   'a' is not defined           no-undef
  Line 4591:21:   'two' is not defined         no-undef
  Line 4677:13:   'i' is not defined           no-undef
  Line 4678:13:   'ascii' is not defined       no-undef
  Line 4680:20:   'i' is not defined           no-undef
  Line 4681:17:   'ascii' is not defined       no-undef
  Line 4681:25:   'ascii' is not defined       no-undef
  Line 4682:41:   'i' is not defined           no-undef
  Line 4683:17:   'i' is not defined           no-undef
  Line 4686:20:   'ascii' is not defined       no-undef
  Line 5827:25:   Unexpected use of 'self'     no-restricted-globals
  Line 5828:19:   Unexpected use of 'self'     no-restricted-globals
  Line 6530:44:   'value' is not defined       no-undef
  Line 6568:65:   'async' is not defined       no-undef
  Line 6568:72:   'sjcl' is not defined        no-undef
  Line 6568:78:   'BigInteger' is not defined  no-undef
  Line 6568:90:   'SRPClient' is not defined   no-undef
  Line 6568:101:  'ES6Promise' is not defined  no-undef
```
  
4. Crypto and Vertx (though this one doesn't make an error message pop-up in the environment)
```
WARNING in ./src/modules/WebSdk/index.js 2445:15-32
Module not found: Error: Can't resolve 'crypto' in 'E:\xampp\htdocs\bjmp\frontend\src\modules\WebSdk'

BREAKING CHANGE: webpack < 5 used to include polyfills for node.js core modules by default.        
This is no longer the case. Verify if you need this module and configure a polyfill for it.        

If you want to include a polyfill, you need to:
        - add a fallback 'resolve.fallback: { "crypto": require.resolve("crypto-browserify") }'    
        - install 'crypto-browserify'
If you don't want to include a polyfill, you can use an empty module like this:
        resolve.fallback: { "crypto": false }

WARNING in ./src/modules/WebSdk/index.js 4602:18-28
Module not found: Error: Can't resolve 'vertx' in 'E:\xampp\htdocs\bjmp\frontend\src\modules\WebSdk'
```

Again, I read most of the issues from other threads and tried doing them but to no avail. If anyone could help I'd appreciate it! Thanks.

a-bronx, 2023-09-25:
Based on the error messages, you attempt to use the library in Node.js code. The library is [for browsers only](https://hidglobal.github.io/digitalpersona-devices/how-to.html).

closed 2023-09-25 by a-bronx

shinomikko8888, 2023-09-26:
> The library is [for browsers only](https://hidglobal.github.io/digitalpersona-devices/how-to.html).

Sorry, but I'm still confused about this part. Do I have to call the `<script>` that calls the WebSdk and the @digitalpersona/devices script in the `index.html` file that is located inside the public folder and program there using JavaScript instead of programming inside of the component? Or do I have to call the `<script>` tag inside the return statement in the component since inside of the return statement you can also call the `<script>` tag, and continue as normal?
My scenario is the user entering an entry that contains data from a person which includes their fingerprint data.

What I want is to get the fingerprint raw file (or anything that could get a unique ID) and compile the data with the other data that has been inputted such as the name, age, etc.
When the user clicks on this button:
[image]

This would pop-up:
[image]

So, in this scenario, where should I call the  `<script>` tag that calls the WebSdk?

a-bronx, 2023-09-26:
> Or do I have to call the <script> tag inside the return statement in the component since inside of the return statement you can also call the <script> tag, and continue as normal?

I do not know do you talk about React components, Angular components, Vue components or some other components.

But irrespective of what framework you use, and what bundler you use, in your final code both the `@digitalpersona/devices` and the `WebSdk` must be loaded into the `index.html` using its regular `<script>` tag, like any other browser script:

```html
<html>
<head></head>
<body>
....
   <script src="bundle-with-the-websdk-and-digitalpersona-devices-inside.js"></script>
</body>
</html>
```

shinomikko8888, 2023-10-19:
I came back to this again, I never resolved the issue because I was focusing on other easier things in my system before this. I found a solution albeit I used an older or probably outdated SDK that worked for the WebAPI example that came with it. The problem is when I copy exactly what you did in [here](https://hidglobal.github.io/digitalpersona-devices/tutorial.html) and [here](https://github.com/hidglobal/digitalpersona-sample-angularjs/blob/2c54be9c09434bdac39298162e4e6ff7316038c7/src/index.ts) for the SDK listed here

I'm still getting the 2nd error.
But if I put it in the src/ folder in my project and call it exactly like in the example. I get the variables error.

The only solution I found (the one from the WebAPI example, not this one) was putting the file in the public folder in my React app, then creating a script tag in a useEffect in React and calling the functions globally like this:

```
/* 
   global selectChangeEvent 
   global readersDropDownPopulate 
   global populatePopUpModal 
   global disableEnable 
   global enableDisableScanQualityDiv 
   global disableEnableExport 
   global FingerprintSdkTest 
   global inputOnStart
   global onStop
   */

//Other code here

useEffect(() => {
    const script = document.createElement('script');
    script.src = '/websdk.js';
    script.async = true;
    script.defer = true;
    const imagediv = document.getElementById('imagediv');
}, []);
```
 It worked, but again, I'm not planning to use that one and use this package along with the authentication and enrollment, and I'm stuck on this part where I can use the constructor FingerprintReader(); and the other parts of the package since in React, when I type 
``
const reader = new FingerprintReader();
//Or anything in the tutorial 
``

...as a test in either my App.js or my index.js inside the src/ folder, it auto imports the FingerprintReader constructor from the node_modules package. What do I have to do since I already imported the modules inside the `index.html` file like this:
`<script src="/modules/WebSdk/index.js" defer></script>`

Do I also have to reference the @digitalpersona/ script inside the `index.html` inside the public folder? I tried to do that but the folder isn't being detected. Or do I have to create a new .js file in the folder, and create something similar to the one in the tutorial there?

Sorry, if I sound a little confusing or oblivious, I'm already at my wits end because this is the only feature left before I'm finished, and its taking a little too much time as a student to learn this. Thanks

a-bronx, 2023-10-19:
> Do I also have to reference the @digitalpersona/ script inside the index.html inside the public folder?

Yes. You can either manually copy one of the `node_modules\@digitalpersona\devices\dist\*.bundles' file into your public folder and add the corresponding `<script>` tag in the `index.html`, or, more typically, setup an automatic bundler (webpack etc) so that it pulls all browser-side libraries referenced in your code from `node_modules`, concatenate them into a JS bundle in the public folder (with dead code elimination, minification etc), and insert the `<script>` tag automatically.

shinomikko8888, 2023-10-21:
Ah, I see, I kind of got it to work, but it functioned similarly to the old one so I might not rewrite the code for implementing it in React since I'm almost near the deadline and I already have something working right now. But thanks for the help and support regardless!!

Also one last thing, though not really related to the devices, more on recognizing the fingerprint (I think this might be on the enrollment or authentication, not sure on what) and the flow of my system:
The flow right now should be:
- The admin creates an entry for a user.
- The form asks for a fingerprint to recognize the user along with other details.
- The user scans their fingerprint and the resulting string (I guess for me, it returns a really long string, because I made it save as an Intermediate file) is saved in the localStorage temporarily
- After submitting the form, a .bin file is generated and saved to a folder for future access, the file will be named by a number of parameters that are made during the submission. That's on me to decide.
- The fingerprint will be used as an alternative for "pseudo-payment" and the matching will be present when the user checks out their items.

My question is, could I use `@digitalpersona/authentication` and/or  `@digitalpersona/enrollment` for matching .bin files? Or does the package use other file types? Currently in the system I have, I can only do 4, and these 4 are the presets in the WebAPI sample which are PNG, RAW, WSQ, and the Intermediate/Feature Set.

a-bronx, 2023-10-23:
Just for a case: you obviously cannot match samples using just a binary comparison. For fingerprint matching you need a specialized fingerprint matching engine which should be able

* to extract fingerprint features ("minutia"s) from the image and organize them in "feature sets" optimal for matching
* to compare two such feature sets estimating similarity of two feature sets (e.g., by calculating a relative "distance" in some abstract space of features, or by some advanced pattern recognition algorithm)
* to make a "match/no match" decision based on some target statistics ("false accept rate", "false reject rate" etc).

The `@digitalpersona/authentication` libray does not do mathcing itself, it just helps to encode and send the samples (captured by the `@digitalpersona/devices`) to the DigitalPersona Authentication Server, which is a commercial HID product. The server has a fingerprint matching engine ("FingerJet") and uses it to compare the sample with the one previously enrolled during user registration (probably using the `@digitalpersona/enrollment`) and stored in the user's ActiveDirectory record. So, both  `@digitalpersona/authentication` and `@digitalpersona/enrollment` has a DigitalPersona Server as a prerequisite. If you do not have one, those libraries will have no use for you, they are just thin web client wrappers.

You can also use DigitalPersona Fingerprint SDK (another HID product) which also includes the FingerJet engine, but does not include the server doing ActiveDirectory integration. With it you'll need to store fingerprint templates inside your own database, and your backend server should call the FingerJet functions to do both the feature extraction and the matching. You'll also need to care about sending collected fingerprint samples from the web browser to your backend.

And finally, you can use any other fingerprint engine of your liking and pass raw images to it. You won't be able to use `Intermediate` format for this, as it is a DigitalPersona proprietary format intended to be used with DigitalPersona FingerJet engine only. You'll need to use your fingerprint engine both to extract fingerprint features from the image, and to compare feature sets of the presented finger with the stored ones for identification.

locent217, 2024-01-09:
@a-bronx Hello sir! Can you help me with this error?
[image]

a-bronx, 2024-01-09:
@locent217 as it says in the error message, you should "externalize" the WebSdk module, i.e. add it to the "external" collection in your Rollup config (and also the the list of "globals"). Something like this:

*rollup.config.js:*
```
export default {
    input: 'dist/index.js',
    external: [ 'WebSdk' ],
    output: {
        ..., 
        globals: {
            'WebSdk': 'WebSdk',
        }
    },
    plugins: [ ... ]
}
```

locent217, 2024-01-14:
@a-bronx I finally fixed the "failed to resolve module specifier "websdk"" error in an unorthodox way, I just commented the 

```// import 'WebSdk';``` in the node_modules/@digitalpersona/devices/dist/es5/devices/websdk/channel.js file.

From what I understand, it read the es5 module first before going to es5 during compile. I am using es6 and Vue3+Vite if it helps. Thanks!

a-bronx, 2024-01-16:
@locent217 Thank you for your observation. Seems like Vite behaves differently from Webpack which I used for bundling my samples. I'll look into the intarction with other bundlers.