hidglobal/digitalpersona-devices #49: sir @a-bronx  it's possible to convert PNG Image to Intermediate sample format or Intermediate to PngImage Format?

rangerisrael, 2023-10-19:
sir @a-bronx  it's possible to convert PNG Image to Intermediate sample format or Intermediate to PngImage Format?

_Originally posted by @rangerisrael in https://github.com/hidglobal/digitalpersona-devices/issues/11#issuecomment-1770154467_

a-bronx, 2023-10-19:
"Intermediate -> PNG" is impossible in principle, because all original image information is lost, only a set of fingerprint features is stored.

"PNG -> Intermediate" requires a call to the DigitalPersona Fingerprint Engine. The library does not provide access to the engine, but you can get DigitalPersona SDK which allows this.

closed 2023-10-19 by a-bronx

rangerisrael, 2023-10-23:
@a-bronx can you give sample how I convert Png to Intermediate?

rangerisrael, 2023-10-23:
@a-bronx what is encoded in Intermediate Sample Format it is base64 / Utf8 / base64 url?

a-bronx, 2023-10-23:
> can you give sample how I convert Png to Intermediate?

Such sample is a part of the DigitalPersona SDK, you need to obtain the SDK first.

> what is encoded in Intermediate Sample Format it is base64 / Utf8 / base64 url?

A fingerprint feature set in a DigitalPersona proprietary format.

rangerisrael, 2023-10-23:
> > can you give sample how I convert Png to Intermediate?
> 
> Such sample is a part of the DigitalPersona SDK, you need to obtain the SDK first.
> 
> > what is encoded in Intermediate Sample Format it is base64 / Utf8 / base64 url?
> 
> A fingerprint feature set in a DigitalPersona proprietary format.

@a-bronx  yes I already implement the sdk for digital persona at I get fingerprint image accurately

rangerisrael, 2023-10-23:
[image]

a-bronx, 2023-10-23:
> I already implement the sdk for digital persona at I get fingerprint image accurately,

I do not understand what do you mean by this? DigitalPersona Fingerprint SDK is a commercial product, which contains a fingerprint engine (a native Windows dynamic library) and distributed by HID Global. If you have purchased the SDK already, then you should have samples of the code with the distributives, as well as contacth of the SDK support team which can help you.

> I just want to know how could convert Png Image to Intermediate

You need to invoke the DigitalPersona Fingerprint Engine functions. You cannot do it in JavaScript., as this is a native API, ant it is out of the scope of the `@digitalpersona/devices` library, so I cannot provide you with samples.