hidglobal/digitalpersona-devices #52: Bug Softare DigitalPersona Lite Client (Connected and Disconnected)

heriira, 2023-12-23:
I am using Digital Persona 4500 and trying to run it on a web browser. Everything runs smoothly, and the fingerprint can be read in the browser up to this point. However, after that, I encountered a bug. My fingerprint device keeps connected and disconnected or turning off and on within a few seconds, and it continues like that. I am using DigitalPersona Lite Client. Does anyone have a solution to my problem? Thank you.

locent217, 2024-01-09:
My HID Device also have this bug when i install the LITE CLIENT. But when I install the other sdk it does not disconnect/connect.

heriira, 2024-01-09:
@locent217 besides the 'LITE CLIENT', which SDK do you use? And after installing another SDK, will the fingerprint still be readable on the website platform?

locent217, 2024-01-10:
The version of DigitalPersona Lite Client i used is the current version which is from the tutorial. I downloaded it last January 8, 2024.

I didnt install any drivers because I do not know where to download it and I think DigitalPersona Lite Client automatically installed a driver for it.

I installed the software on to my two laptops and still the device reconnects every like 1-2 minutes.

It just reconnects every 1-2 minutes even in idle.

heriira, 2024-01-10:
@a-bronx 

1. The version of DigitalPersona Lite Client I am using is V 4.0.0.925 from the source: [link to the source](https://www.crossmatch.com/AltusFiles/AltusLite/digitalPersonaClient.Setup64.exe).

2. I did not install any additional drivers because I couldn't find them and thought that the lite client already covers other drivers. Please let me know if there are other drivers that need to be installed along with the lite client.

3. I conducted tests on 2 different laptop devices and experienced similar issues.

4. Yes, I have two fingerprint reader devices. The first one is a device that still uses the DigitalPersona logo, and the second one is a device with the HID logo. Both of them are experiencing similar issues.

5. After I retested, it turns out that even though I haven't opened the browser (Idle), the fingerprint device still reconnects every 1-2 minutes, and it can even be faster.

a-bronx, 2024-01-10:
Can you try v4.1? You can download it [here](https://crossmatch.hid.gl/lite-client/).

heriira, 2024-01-10:
@a-bronx 
Finally, I found a solution to my issue. I tried the latest version from the provided link, and the Lite Client is running smoothly with no continuous reconnect bugs. Thank you very much

a-bronx, 2024-01-10:
LiteClient download links in documentation are updated.

closed 2024-01-10 by a-bronx