hidglobal/digitalpersona-devices #53: Failed to resolve module specifier "WebSdk"

locent217, 2024-01-14:
Hi! I followed the tutorial [Tutorial](https://hidglobal.github.io/digitalpersona-devices/tutorial.html). And I get this error.

I am using vue3 with vite.

Console error: "Uncaught (in promise) TypeError: Failed to resolve module specifier "WebSdk". Relative references must start with either "/", "./", or "../".

a-bronx, 2024-01-14:
I'm not familiar with Vite. Please use Vite's documentation to find out how to reference global UMD modules like WebSdk, and add how to add their typings.

closed 2024-01-14 by a-bronx

JerryA1, 2025-05-09:
> Hi! I followed the tutorial [Tutorial](https://hidglobal.github.io/digitalpersona-devices/tutorial.html). And I get this error.
>
> I am using vue3 with vite.
>
> Console error: "Uncaught (in promise) [...]

Have you solved this? I have this error in my build, everything is fine in development, but when I try to generate the build it fails, I'm using react and vite

a-bronx, 2025-05-09:
You can try using a `@digitalpersona\websdk` package now. I haven't updated this package yet, but there is a very similar `@digitalpersona\fingerprint` package, which supports only fingerprints, or `@digitalpersona\card` which supports only cards.