hidglobal/digitalpersona-devices #55: onDataReceivedTxt error

locent217, 2024-01-16:
[image]

I have the following error. It occurs after I re-start the fingerprint acquisition. Can you help me with the possible cause of it?

a-bronx, 2024-01-16:
The error 0x80070057 is a Windows HRESULT which translates to a message "The parameter is incorrect." This message comes from the "native" part (Lite CLient or fingerprint API or driver) and a possible cause may be an invalid parameter passed into a method like `startAcquisition()` or `stopAcquisition()`

locent217, 2024-01-17:
@a-bronx What might be my mistake why I am having this warning? I followed the tutorial so far. Thank you in advance

a-bronx, 2024-01-17:
Can you provide access to your project's repository or to a repository with a minimal sample reproducing the issue?

locent217, 2024-01-19:
Hello @a-bronx , I managed to fix it by simply using onBeforeUnmount() in vue3. Basically it is an equivalent of delete in the tutorial. thank you!

closed 2024-01-19 by locent217