hidglobal/digitalpersona-devices #59: converted output from b64UrlTo64 to WSQ

wanfahdiva, 2024-03-12:
Is it possible to convert the output from b64UrlTo64 to WSQ format? so I tried this SDK using the example repo from https://github.com/ialeke/React-Fingerpint-Reader, and from this repo it only gives base64 output. is there another way to convert the output from base64 to WSQ format?

a-bronx, 2024-03-12:
Supported sample formats [are following](https://github.com/hidglobal/digitalpersona-devices/blob/859f9b2e8557272d88c237ea844dc1fd7866b5e0/src/devices/fingerprints/sample.ts#L4):

```ts
export enum SampleFormat {
    /** A raw fingerprint image (bitmap). */
    Raw = 1,
    /** A fingerprint feature set in a DigitalPersona proprietary format. To use with DigitalPersona fingerprint matching engine only. */
    Intermediate = 2,
    /** A fingerprint image compressed using Wavelet Scalar Quantization (WSQ) algotithm. */
    Compressed = 3,
    /** A fingerprint image in a Portable Network Graphics (PNG) format. */
    PngImage = 5,
}
```
In you code, change `this.sdk.startAcquisition(Fingerprint.SampleFormat.PngImage)` to `this.sdk.startAcquisition(Fingerprint.SampleFormat.Compressed)`, and you'll get a Base64Url-encoded WSQ image.

closed 2024-03-12 by a-bronx

wanfahdiva, 2024-03-20:
thanks for the answer mr @a-bronx, but i have onemore question. how to display it on html image or save as file .wsq? is it possible?

a-bronx, 2024-03-20:
> how to display it on html image

As far as I know, browsers do not support WSQ images, there is even no MIME type for it. You can try MIME types for JPEG200 (`image/jp2`, `image/jpx`, `image/jpm`) which is also uses DWT, but I doubt it will work. 

So for display purposes you may need to stick to PNG.

> save as file .wsq?

Same as any other file, but as there is no known MIME type for WSQ, you'll probably need to save it as "unknown" file format, using MIME type like `application-octet-stream` (standard) or `application/unknown` (nonstandard) or just empty MIME type (also standard).

wanfahdiva, 2024-03-24:
thank you for the answer sir @a-bronx